openpnp/openpnp #1423: Programmatic editing of machine.xml

surabibio, 2022-05-28:
# Unwarranted Addition of Bottom Vision with programmatic loading of machine.xml
## The Issue
I am trying to implement feeder groups through scripting in java. This requires loading and saving machine.xml through scripting. I find that every call to Configuration.get().load(); increments the Bottom Vision

## Steps to Reproduce
On a fresh installation of OpenPnP create this java script file as below and run the script

```
import org.openpnp.model.Configuration;
Configuration.get().load();

```

## Expected Result
No change anywhere

## Actual Result
Vision > Bottom Vision keeps getting added for every call.

markmaker, 2022-05-29:
Sorry I don' know what you mean by "Bottom Vision keeps getting added for every call" or "increments the Bottom Vision".

Perhaps attach a `machine.xml` before and after? Or add a screenshot?

_Mark

surabibio, 2022-05-29:
Mark, I don't think this way of updating the live machine from a machine.xml file is a good idea. I will try to add only feeders from an xml file to my script. (Please have a look into the script pull request)

ian-arkver, 2022-05-29:
Reproduced here with latest test branch and no initial .openpnp2 dir (i.e. defaults used). Running the config.get.load script adds another instance of
```
         <part-alignment class="org.openpnp.machine.reference.vision.ReferenceBottomVision" bottom-vision-id="BVS_Stock" enabled="false" pre-rotate="false" max-vision-passes="3" max-angular-offset="10.0" test-alignment-angle="0.0">
            <max-linear-offset value="1.0" units="Millimeters"/>
         </part-alignment>
```
every time.

ian-arkver, 2022-05-29:
I took a look at this since it feels like a bug, even if reloading the machine.xml by script is a "Bad Idea". It's only ReferenceBottomVision that's getting recreated, not ReferenceFiducialLocator, even though the two classes share some code around configuration done callback and part migration.

I added a couple of traces to the configurationComplete override in BV...
```
             public void configurationComplete(Configuration configuration) throws Exception {
+                Logger.trace("CfgComplete: PartAlignments={}", configuration.getMachine().getPartAlignments());
                 migratePartSettings(configuration);
+                Logger.trace("MigratePart: PartAlignments={}", configuration.getMachine().getPartAlignments());
                 if (bottomVisionSettings == null) {
```

The log looks like this - one call at program startup, then five (5!) calls at the time of the script call. The class list is remade with the new RBV and new IDs between the 2nd and 3rd calls.

```
2022-05-29 22:17:19.164 ReferenceBottomVision TRACE: CfgComplete: PartAlignments=[org.openpnp.machine.reference.vision.ReferenceBottomVision@4584776f]
2022-05-29 22:17:19.172 ReferenceBottomVision TRACE: MigratePart: PartAlignments=[org.openpnp.machine.reference.vision.ReferenceBottomVision@4584776f]
2022-05-29 22:17:19.262 Main INFO: Bienvenue, Bienvenido, Willkommen, Hello, Namaskar, Welkom, Bonjour to OpenPnP version 2022-05-29_21-16-58.34e6525.
2022-05-29 22:17:19.263 Scripting TRACE: Scripting.on Startup
2022-05-29 22:17:19.275 CameraView DEBUG: Failed to load camera specific reticle, checking default.
2022-05-29 22:17:19.275 CameraView DEBUG: No reticle preference found.
2022-05-29 22:17:19.690 AbstractBroadcastingCamera TRACE: Camera Top thread 40 started.
2022-05-29 22:17:19.693 CameraView DEBUG: Failed to load camera specific reticle, checking default.
2022-05-29 22:17:19.694 CameraView DEBUG: No reticle preference found.
2022-05-29 22:17:19.694 AbstractBroadcastingCamera TRACE: Camera Bottom thread 42 started.
2022-05-29 22:17:33.672 ReferenceBottomVision TRACE: CfgComplete: PartAlignments=[org.openpnp.machine.reference.vision.ReferenceBottomVision@4584776f]
2022-05-29 22:17:33.678 ReferenceBottomVision TRACE: MigratePart: PartAlignments=[org.openpnp.machine.reference.vision.ReferenceBottomVision@4584776f]
2022-05-29 22:17:33.710 ReferenceBottomVision TRACE: CfgComplete: PartAlignments=[org.openpnp.machine.reference.vision.ReferenceBottomVision@4584776f]
2022-05-29 22:17:33.716 ReferenceBottomVision TRACE: MigratePart: PartAlignments=[org.openpnp.machine.reference.vision.ReferenceBottomVision@4584776f]
2022-05-29 22:17:33.741 ReferenceBottomVision TRACE: CfgComplete: PartAlignments=[org.openpnp.machine.reference.vision.ReferenceBottomVision@dc0359c, org.openpnp.machine.reference.vision.ReferenceBottomVision@4639a0fe]
2022-05-29 22:17:33.746 ReferenceBottomVision TRACE: MigratePart: PartAlignments=[org.openpnp.machine.reference.vision.ReferenceBottomVision@dc0359c, org.openpnp.machine.reference.vision.ReferenceBottomVision@4639a0fe]
2022-05-29 22:17:33.753 ReferenceBottomVision TRACE: CfgComplete: PartAlignments=[org.openpnp.machine.reference.vision.ReferenceBottomVision@dc0359c, org.openpnp.machine.reference.vision.ReferenceBottomVision@4639a0fe]
2022-05-29 22:17:33.758 ReferenceBottomVision TRACE: MigratePart: PartAlignments=[org.openpnp.machine.reference.vision.ReferenceBottomVision@dc0359c, org.openpnp.machine.reference.vision.ReferenceBottomVision@4639a0fe]
2022-05-29 22:17:33.762 ReferenceBottomVision TRACE: CfgComplete: PartAlignments=[org.openpnp.machine.reference.vision.ReferenceBottomVision@dc0359c, org.openpnp.machine.reference.vision.ReferenceBottomVision@4639a0fe]
2022-05-29 22:17:33.767 ReferenceBottomVision TRACE: MigratePart: PartAlignments=[org.openpnp.machine.reference.vision.ReferenceBottomVision@dc0359c, org.openpnp.machine.reference.vision.ReferenceBottomVision@4639a0fe]
2022-05-29 22:17:33.776 CameraView DEBUG: Failed to load camera specific reticle, checking default.
2022-05-29 22:17:33.776 CameraView DEBUG: No reticle preference found.
2022-05-29 22:17:34.156 AbstractBroadcastingCamera TRACE: Camera Top thread 45 started.
```
Not sure where to look next and out of time for this evening. Hopefully the above helps narrow it down.

surabibio, 2022-05-30:
Thanks Ian for giving life to this approach, which I had almost given up. Updating portion of machine.xml could open new possibilities to the user - like trying different vision pipelines apart from feeder management.  Also the machine gets switched off and locations get zeroed after config.get.load. Can this also be avoided?

markmaker, 2022-05-30:
Hi @ian-arkver, @surabibio,

I don't understand, why one needs to reload the configuration, to change stuff from a script. You can arguably _more easily_ change it directly in the Java object tree. After all the XML serialization is driven by reflection. i.e. it is more or less a 1:1 image of the object tree (and vice versa). 

I won't claim to have understood 100% of the ramifications of reloading the configuration, but as far as I understand, this will never work. While it is possible to reload the configuration (it is done in unit tests), there is AFAIK only one GUI instance (rooted in `MainFrame`) supported, that I assume will **not** be recreated, so any Java objects from the old configuration that are referenced from components, actions, listeners (lambdas, generally) will still be wired up. Some of the newly loaded objects will perhaps be properly installed in the GUI, but for most single-instance objects this is usually not the case, and the wrong (old) object will still be wired. So you'll probably get very strange behavior, including lost updates when saving the new configuration. Sometimes duplicates will appear, especially in listeners (which is apparent in this case). 

Listeners are especially problematic, because they are not [weak-referenced](https://en.wikipedia.org/wiki/Weak_reference) in general, i.e. they will keep the old objects alive, the Garbage Collector will not clean it up. Sometimes you will probably have old objects still being zombie-functional in the background and disrupting operation (like both the old and new object reacting to events and actions, the old driver still connected to the controller, the old camera still streaming etc.).  

If (after reading the first paragraph) you still want to do this, I suggest you fully restart a new OpenPnP process from the script and then immediately stop the current one (without saving). Perhaps you would need to call a shell-script with a delay before the OpenPnP re-launch, so the new process does not stumble over file locks etc. from the old.

_Mark

ian-arkver, 2022-05-30:
Thanks Mark. It sounds like there are many other reasons why reloading the config in a script is a "Very Bad Idea", so this one isn't a particularly high priority. @surabibio should probably find another approach using other scripting hooks.

I'm back at $DAY_JOB now, so won't have time to go chasing down a dangling strong object reference in some list somewhere (if that's what it is).

All the best, Ian

ian-arkver, 2022-05-30:
ps: Maybe there's an argument for splitting machine.xml to separate out the feeder config. Feeders will change much more frequently and dynamically than the base machine. That's a separate discussion for the group though.

markmaker, 2022-05-30:
This is all not really addressing the real underlying issue, which is: OpenPnP should have logical feeder groups and mounting/unmounting. For BlindsFeeder arrays, other slate based designs, reel trolleys, or even single feeder slots. 

See #943, but apart from just solving the location translation issue, there must be a mechanism to switch whole groups "unmounted" (which is _not_ the same as "feeder disabled"). 

In general, I doubt having multiple XML files is a good idea, because these feeders would then become stale. The part connection is no longer there, i.e. you lose the ability to track a part to its group and initiate the "mounting". And if there are functional updates in OpenPnP, the migration handlers will not properly run (or at least it would become much more complicated to sort-of retroactively run them when mounted). 

_Mark

ian-arkver, 2022-05-30:
Yes, I agree, doing it properly within OpenPnP is the best way. Managing feeder groups by switching in and out separate xml files would be hacky.

Re "track a part to its group"... Surely that's not a 1:1 relationship? Parts can be on multiple trolleys/slates as different jobs demand - or not currently assigned to any feeder bank.

As Jason and others said in #943, the community needs a discussion in the forum to decide what's actually needed.

Regards, Ian.

markmaker, 2022-05-30:
/> Surely that's not a 1:1 relationship?/

Correct, 1:1 was not implied, just to find a slate, slot mountable 
feeder, trolley etc. that contains this part (among many possible 
others, which all can be already mounted or unmounted). /How/, would 
have to be discussed.

_Mark

surabibio, 2022-06-03:
Hi Mark and Ian,
With your inputs I have modified the approach to saving and loading feeders. This time I am using machine.removeFeeder() and machine.addFeeder(). Please have a look at the PR submitted in the openpnp/scripts repository and do the needful.
Ravi

markmaker, 2022-09-12:
Moved to https://github.com/openpnp/scripts/pull/2

closed 2022-09-12 by markmaker