openpnp/openpnp #1432: Example not work

Uchaplin, 2022-06-28:
# Read This First
Issues must only be opened once they are confirmed.
Please [visit the discussion group](https://groups.google.com/group/openpnp) first, so bugs or missing features can be separated from other problems.

To submit an issue for OpenPnP you must use this template or it will be deleted. Please fill out either the Problem Report or Feature Request section below and delete the other section.

-----------------------------------------------------------------------

# Problem Report
## Describe the Issue
Please describe the issue in as much detail as possible. Without a good description it's impossible for developers to understand the problem.

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. Start OpenPnP
2. Click the Power On button.
3. Jog the machine 10mm to the right.
4. The machine fails to jog and shows an error.

## Expected Result
What did you expect to happen? Be descriptive.

## Actual Result
What actually happened? Be descriptive.

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
The more information you can provide about the issue the more likely it is that a developer will be able to respond.

Whenever possible, please use the Menu -> Help -> Submit Diagnostics feature and include the URL here. Submit Diagnostics makes it easy to include log files, screen capture, debug images, configuration files, etc.

-----------------------------------------------------------------------

# Feature Request
## Describe the Feature
What do you want it to do? Be as descriptive as possible.

## How Will It Look?
If the feature will include changes to the user interface, how would they look? Include drawings or mockups if possible.

## Why Do We Need It?
1. What problem does it solve?
2. Is it useful for everyone, or does it only solve a problem for your machine?

## Provide Examples
Is this something that exists in other pick and place software? Provide examples and references if possible.

Uchaplin, 2022-06-28:
I tried examples in Windows 7 and it not work. I receive message "Attribute version does not a match in class.org.openpnp.models.Job  at line 3"

markmaker, 2022-06-28:
_Please [visit the discussion group](https://groups.google.com/group/openpnp) first, so bugs or missing features can be separated from other problems._

_Mark

markmaker, 2022-09-12:
Seems abandoned.

closed 2022-09-12 by markmaker