openpnp/openpnp #1437: Raspberry Pi camera is not working under 64-bit version of Raspberry Pi OS

dronecz, 2022-07-09:
# Problem Report
## Describe the Issue
I tested Raspberry Pi 4 with Raspberry Pi v2 camera and OpenPnP running on 64-bit version of Raspberry Pi OS and I can add new OpenPnpCaptureCamera, but I´m not able to use this camera. I can see multiple `bcm2835-isp` devices, but all have `0 x 0 px` resolution in all formats (see images).

[image]
[image]

I also tried OpenCVCamera, but I can not see any picture from camera and I can see `ReferenceCamera TRACE: Camera OpenCvCamera failed to return an image. Retrying.` in the log. If I try `libcamera-hello` command, I can see that camera is working. 

```
Made X/EGL preview window
[0:11:36.705114409] [2988]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3700-f30ad033
[0:11:36.784319123] [2991]  WARN RPI raspberrypi.cpp:1252 Mismatch between Unicam and CamHelper for embedded data usage!
[0:11:36.785011579] [2991]  INFO RPI raspberrypi.cpp:1368 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media3 and ISP device /dev/media0
[0:11:36.785957198] [2988]  INFO Camera camera.cpp:1029 configuring streams: (0) 1640x1232-YUV420
[0:11:36.786395212] [2991]  INFO RPI raspberrypi.cpp:759 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 1640x1232-SBGGR10_1X10 - Selected unicam format: 1640x1232-pBAA
```

I also tried same HW setup under 32-bit version of the system and there is camera normaly working via OpenCVCamera.
 

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. Install OpenPnP on Raspberry Pi OS x64 with RPi camera v2 attached
2. Get openpnp-capture-java-0.0.23.jar and place it in /opt/openpnp/lib folder
3. Start OpenPnP
4. Try to add new OpenPnpCaptureCamera or OpenCVCamera.

## Expected Result
I expected to see some image. 

## Actual Result
I can not use RPi camera v2 with OpenPnP. 

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
I´m unable to submit diagnostics so here is [log](https://github.com/openpnp/openpnp/files/9078033/OpenPnP_log.txt) file.

markmaker, 2022-07-10:
Hi @dronecz, 

I guess you need to open his issue at [openpnp/openpnp-capture](https://github.com/openpnp/openpnp-capture/issues) (different repo).

See also here, and perhaps contact @brainstorm:
https://github.com/openpnp/openpnp-capture/pull/47

_Mark

brainstorm, 2022-07-14:
Thanks Mark for the @-mention,

Hello again @dronecz, long time since we talked about this ;)

I have no experience with those cameras *and* openpnp, so far I've only played with USB ones and [openpnp-kiosk](github.com/CCHS-Melbourne/openpnp-kiosk). I'll give a try with a spare camera I have (Arducam) and see if it gives much trouble with Jetson Nano and that CSI connector... That's all I can do for now as I don't have your exact hardware combination :-S

I have a newborn to take care of now, so my time is rather limited ;)

dronecz, 2022-07-14:
Thanks for reply and congrats to your new little happiness ❤. 

As @markmaker pointed, this maybe problem in `openpnp-capture` part, but I´m not sure about that as I do not know how listing of the divices work under the hood of openpnp. 

I´m willing to test what I can, but my knowledge of linux systems is bit limited. Maybe I could test some files from [openpnp-capture](https://github.com/openpnp/openpnp-capture/releases/download/v0.0.23/openpnp-capture-test-ubuntu-latest-aarch64) releases, but I do not know how to execute them. 

I understand that you will have now less time, as kids need lot of attention, but I feel like this is close to finish line with all that hard work you already did it in meantioned PR [#47](https://github.com/openpnp/openpnp-capture/pull/47). 

You can ping me at Discord if you want 😉. Thanks

darkomenz, 2022-11-21:
I am also seeing this issue on the RPI CM4 so I opened a ticket at https://github.com/openpnp/openpnp-capture/issues/60