openpnp/openpnp #1451: ReferenceRotatedTrayFeeder not picking from correct location

infinitemachinery, 2022-07-23:
[machinesettings.zip](https://github.com/openpnp/openpnp/files/9174242/machinesettings.zip)
[OpenPnP.log](https://github.com/openpnp/openpnp/files/9174244/OpenPnP.log)
# Read This First
Issues must only be opened once they are confirmed.
Please [visit the discussion group](https://groups.google.com/group/openpnp) first, so bugs or missing features can be separated from other problems.

To submit an issue for OpenPnP you must use this template or it will be deleted. Please fill out either the Problem Report or Feature Request section below and delete the other section.

-----------------------------------------------------------------------

# Problem Report
## Describe the Issue
I have a ReferenceRotatedTrayFeeder defined, 2 rows and 20 columns. I have defined the beginning position, last position on row 1 and last position on row 2, as required by the tray definition.

The problem I'm having is that when I pick the first part from the tray, rather than picking from row 1 from column 1 it picks from column 1 row 2. I have the current feed count set to 0.

[image]

## Steps to Reproduce

1. Start OpenPnP
2. Connect to machine and home.
3. Load a job
4. In the Feeders tab create a ReferenceRotatedTrayFeeder with the necessary rows and columns. Establish the A, B and C points of tray using camera vision.
5. Establish the part for the feeder.
6. Set Feed Count to 0 
7. Place a part in position row 1, column 1 of tray.
8. Perform feed and pick on the feeder, or alternatively place a part from feeder using the job.

## Expected Result
Part is picked from beginning row and column of the feeder, row 1, column 1.

## Actual Result
Part is picked from row 2, column 1 of feeder.

-----------------------------------------------------------------------

tonyluken, 2023-08-19:
@infinitemachinery,

Can you try the latest test branch? This problem should be fixed - more info available in the PR mentioned above.

Tony

tonyluken, 2023-09-13:
Since I haven't heard anything, I'm going to close this issue. If you are still having problems, you can always reopen it or open a new issue.

closed 2023-09-13 by tonyluken