openpnp/openpnp #1453: Feature Request: Semi-automated strip feeder reload/re-locate

qwertymodo, 2022-07-24:
# Feature Request
## Describe the Feature
For speed reasons, and also because of rather frequent vision issues where the top camera randomly can't detect tape holes in certain areas of strip feeders, I have started running my strip feeders with vision disabled.  This works much more reliably, and much faster, but it does come with one large drawback, which is that reloading the feeders requires manually re-capturing the first and second hole locations for every single strip feeder every time I reload, as the tape may have moved forward or backwards along the strip, compared to where it was previously.  It should be possible to utilize the top vision to just attempt to automatically re-capture the first and second holes, based on their previous locations, which would then allow the strip feeder to be used without vision on every pick.

## How Will It Look?
On the GUI side of things, I have two ideas, one or both of which could be implemented to trigger this automatic re-capturing.  The first would simply be a button, located on the strip feeder configuration panel, which could be clicked to trigger the action.  The other option would be a checkbox or drop-down menu to automatically trigger the action prior to the first pick attempt either on a per-job basis, or else when the feed count is 0 (indicating the strip feeder has been reset, and is in need of re-capturing).  In any case, however it is triggered, the following actions would occur:

1. Move the camera to the previously stored location for the first hole.
2. Use the hole location vision pipeline to locate the nearest hole to the current location.
3. Store that nearest hole as the new first hole location.
4. Calculate the position offset between the old first and second holes, and apply that offset to the new first hole in order to calculate the expected location of the new second hole.
5. Move the camera to the expected location of the second hole, and use the camera to attempt to locate the second hole.
6. If both holes are located successfully, store their updated positions.
6a. If either hole fails to be located, do not store the updated position, and emit an error.

## Why Do We Need It?
This would greatly simplify and speed up the process of reloading strip feeders, without the need to enable vision for every pick.  This is basically the exact workflow that I am currently executing manually for every single strip feeder, every time I reload them, and when you find yourself manually doing the same thing over and over, that's always a good sign that you should look into automating it.

## Provide Examples
N/A

markmaker, 2022-07-27:
Hi @qwertymodo ,

You should note that you can do all that and much more using the [BlindsFeeder](https://github.com/openpnp/openpnp/wiki/BlindsFeeder). 

Conversely, strip feeders are really thought for "short term" use, because once the parts are used up, you can't reload them (assuming you have simply taped them down). So the notion of removing and then restoring feeders from/to the machine does not normally enter the picture.

The BlindsFeeder adds reload capability using an extremely minimal (and relatively fast) 3D print. It is from the start intended for relocation, having a cover to keep parts from falling out. It uses a single set of corner fiducials to vision-calibrate a whole _array_ of feeders. So the amortized _per part_ vision operation overhead is very small. 

If you don't want to use the [BlindsFeeder](https://github.com/openpnp/openpnp/wiki/BlindsFeeder), then I would like to know why, so maybe it could be improved in the future. 😁

_Mark

qwertymodo, 2022-07-27:
> Hi @qwertymodo ,
> 
> Conversely, strip feeders are really thought for "short term" use, because once the parts are used up, you can't reload them (assuming you have simply taped them down). So the notion of removing and then restoring feeders from/to the machine does not normally enter the picture.

Perhaps this was a bit of miscommunication on my part.  When I say "relocate" I don't mean moving the feeder around on the machine, I'm talking about a strip feeder that is rigidly attached to the machine, but once the strip runs out and you put a new strip in, it doesn't have any way of ensuring that the new strip is exactly in the same place, it can slide forward and backward within the feeder, so you have to locate the new holes again, which is what I meant by relocate.

Yes, I'm aware that BlindsFeeders support a lot of improvements over the strip feeders, but it's also true that a lot of new users will start with strip feeders, and this would be a fairly simple way to improve quality of life for those users.  As to why I'm not using BlindsFeeders myself, I tried them and had difficulty getting the tolerances down on the sliding mechanism, so I just went back to strips, and now I'm moving on completely to ratcheting lever feeders.  I just wanted to leave this as feedback on my user experience.  I looked into trying to implement it myself, but it went a bit beyond what I was able to figure out myself, and now that I'm moving away from strip feeders, it's not something I'm able to spend more time on.