openpnp/openpnp #1469: Bug / 'Next Placement' button causing exception on missing part.

noeldum, 2022-08-13:
# Problem Report
## Describe the Issue
When using the 'next placement' button to a missing part in the job an error appears. It looks like this is unmanaged.
[image]

## Steps to Reproduce
* Load the attached job. x.job.xml
* Move the machine to the part just before the missing part. (RD1)
* Hit the 'next placement' button.
* Enjoy the exception on the screen.

## Expected Result
Machine should probably move to the position regardless of the part being defined or not. Or maybe skip it at least.

## Actual Result
The error below appears
[image]

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
see attached x.job.xml file to reproduce it easily.
[x.zip](https://github.com/openpnp/openpnp/files/9331188/x.zip)

markmaker, 2022-08-13:
@noeldum, thanks for reporting. This will be fixed in the upcoming roundup.

markmaker, 2022-09-12:
Forgot to report, this was fixed in #1472.

closed 2022-09-12 by markmaker

noeldum, 2022-09-15:
Can do but will take a while. We are busy with production at the moment. Also my openpnp machine has been temporarily decommissioned because we are getting a new pick and place. I'll surely give this a try when possible.