openpnp/openpnp #1470: Bug / Board Enable tick box setting all boards corrdinates same as first board.

noeldum, 2022-08-13:
**I would say this is a rather important problem.**

# Problem Report
## Describe the Issue
When using panels with multiple boards and individual coordinates for each board if we hit the enable button for the first board all coordinates get reset to the same value as the fist board.
Note this is a very frustrating problem. If I have spent a fair bit of time to adjust each board coordinates I clearly do not want to see them non-reversibly flushed away with a single click.

## Steps to Reproduce
* Just make a panel with say 4 boards.
* Adjust some coordinates for each board (maybe just change the rotation by 0.1 to make it easy). In practice I usually do that with the 'multiple placement' button but for reproducing purpose here there is no need for that.
* Click on the enable tick box for the first board.
* Notice all you hard work has been wiped away replacing all boards coordinate with the first one.

[image]
[image]
[image]

## Expected Result
Toggling the enable tick box of the first board should have no effect on any board coordinates.

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
None. Protocol to reproduce the problem is simple enough. Let me know if you need anything.

markmaker, 2022-08-13:
@tonyluken,
you said earlier that you are working on this, right?

Am I right to assume that the _underlying_ problem is the same one as reported [here](https://groups.google.com/g/openpnp/c/ovc3hja6cak/m/dOd30BDvAAAJ);

> Another issue that I noticed is if you are placing the components for example on two boards, and you ran out of the components in the middle of the job, so in case you put one component on one board and you ran out of it, you will get an empty "red" component like "missing feeder" but you will get a check mark as the component was "Placed" on both boards, but in reality, it was only placed on one board because as I see the component list is shared between all panel boards, it does not "multiply" the placement file where you would be able to see and track every component on each board in a panel.

I.e. the panelizing function does not properly deep copy independent instances of boards and their placements?

_Mark

tonyluken, 2022-08-13:
@markmaker,

Yes, I'll take care of this as part of my panelization upgrades.  Note there are several other open issues regarding panelization that I also plan on addressing: [#1464 ](https://github.com/openpnp/openpnp/issues/1464), [#1444 ](https://github.com/openpnp/openpnp/issues/1444), [#1326 ](https://github.com/openpnp/openpnp/issues/1326), [#949 ](https://github.com/openpnp/openpnp/issues/949), [#918 ](https://github.com/openpnp/openpnp/issues/918), [#881 ](https://github.com/openpnp/openpnp/issues/881), [#823 ](https://github.com/openpnp/openpnp/issues/823), [#501 ](https://github.com/openpnp/openpnp/issues/501), and partially [#813 ](https://github.com/openpnp/openpnp/issues/813)

Tony

noeldum, 2022-08-13:
You are very welcome Tony. Thanks a lot.
Panelisation does need to be brought up to speed with the rest I think.

There is another issue that I am about to file but maybe it will be already taken care of. I'll summarise it below.
" When stopping a panelised job with the red square button the remaining unplaced parts identical to the last pick part are marked as 'placed' although they are physically not."

If you think that's very different to the rest I'll open another issue for it.

tonyluken, 2022-08-13:
@noeldum,
I think that is probably the same issue as reported in #823

GeoffState, 2022-10-21:
I noticed this while addressing #1444 and know why it happens. I am currently working on a solution.
I believe the bug occurs because the code is written such that should you add a board with the same name as one already on the boards panel, the file is NOT loaded but assigned a copy of the board with the same name. You have noticed the effect this has.

GeoffState, 2022-10-24:
I am preparing a pull request that solves this problem. 
Expected Result above, is achieved

tonyluken, 2022-10-24:
@GeoffState, can you hold off on this?  I've already implemented a fix for this plus a lot more.  There's no use in us both working the same issue.

GeoffState, 2022-10-24:
OK, #1444 and #1470 are yours @tonyluken, thanks.

noeldum, 2023-02-26:
Our openpnp machine is back running again after a long pause. I installed the latest test package.
A great thank you to all who contributed to the new design of the board/panel interaction.
It is way way better. Rather intuitive and logical.  Good job!

The bug I reported in here is fixed from the little I have tested. We will be producing tomorrow with this test version and will let you know any comment if any.

I don't know how you guys manages the bug reports but I guess this once can be closed.

markmaker, 2023-06-07:
I guess after that, no news is good news! :smile: 

Thanks again, @tonyluken.