openpnp/openpnp #1477: Anyone can edit the wiki?

morganrallen, 2022-09-02:
Someone added a tracking link to the wiki and I was able to undo it. This probably isn't ideal.

https://github.com/openpnp/openpnp/wiki/Setup-and-Calibration:-Machine-Setup/_compare/819e7cc909b81fc8f7a43dbcd40ae6c086d0493f...b5f6c8cbef4bb4504e009ddc2b5af3c4c2b6e09d

vonnieda, 2022-09-02:
Thank you for finding that and making the change. I'll check Git history and see if any other edits were made. The user that made the edit appears to no longer exist. 

I've always felt that anyone being able to edit the Wiki was the best way to let people get involved, but this is a real bummer. I'll give it some thought.

vonnieda, 2022-09-02:
I've found several more evil edits. I'm reverting them now and I've locked down the Wiki to people with commit access for now. CC @markmaker

vonnieda, 2022-09-02:
Alright, I think I've cleaned them all up. Thanks again @morganrallen for catching this and fixing it. I'm leaving the Wiki locked down for now until there's a good reason not to.

closed 2022-09-02 by vonnieda