openpnp/openpnp #1488: Typo in Named CSV Import

GeoffState, 2022-09-21:
# Read This First
Issues must only be opened once they are confirmed.
Please [visit the discussion group](https://groups.google.com/group/openpnp) first, so bugs or missing features can be separated from other problems.

To submit an issue for OpenPnP you must use this template or it will be deleted. Please fill out either the Problem Report or Feature Request section below and delete the other section.

-----------------------------------------------------------------------

# Problem Report
## Describe the Issue
Typo in Named CSV Import

## Steps to Reproduce
1. Open File > Import Board > Named CSV
2. Error occurs in the first word on the Files pane
3. reads Centeroid File (CSV)

## Expected Result
Should read Centroid File (CSV)

## Actual Result
reads Centeroid File (CSV)

## Screen Capture
[image]

-----------------------------------------------------------------------

GeoffState, 2022-09-21:
**TO FIX**
**File:** NamedCSVImporter.java
**Line:** 403: JLabel lblTopFilemnt = new JLabel("Cen**ter**oid File (.csv)"); 
Centeroid should read Cen**tr**oid

markmaker, 2022-09-22:
Thanks, @GeoffState, done in #1489

closed 2022-09-22 by markmaker

markmaker, 2022-09-22:
Now available in the `testing` version. I don't think it is important enough to cherry-pick to the `regular` version.