openpnp/openpnp #1495: Openpnp trademark squatted

listff, 2022-11-20:
Openpnp is registered as a trademark in China,
[image]

this is his youtube channel https://www.youtube.com/channel/UC0HdFo97MsES_ZxW6iCv4Rw/videos
his website https://www.openpnp.cn/

https://www.openpnp.cn/archives/3030
On this page you can see that his company name is 河北擎展智能科技有限公司

This is the company that squatted the openpnp trade name

Trademark Search Website wcjs.sbj.cnipa.gov.cn

vonnieda, 2022-11-21:
Thank you for this @listff.

listff, 2022-11-22:
He has a history of many patent infringements, he manufactures feeder imitations of feederfinger patents and applies for patents in China @vonnieda

listff, 2022-11-25:
Hi vonnieda, is there any progress in protecting the openpnp trademark? @vonnieda

4044ever, 2022-11-25:
Once you have Chinese trademark squatter that is basically it. From what I know, it's gone forever. 
Actually I was planning to get something from Alan0131, now I reconsider.

Alan-0131, 2022-12-06:
大家好我是Alan。我来自中国河北省。 2020年6月我做了第一台兼容openpnp的自动贴片机。从那时起，我就知道这是一个非常有前途的项目，也是一个伟大的项目。 于是我成立了公司，建了网站，设计生产贴片机。 经过两年多的时间，我们总结了自己的不足，取得了一定的成绩。 不久前，我们正式推出了第二代产品。

在中国，我申请注册了openpnp的商标。总共4类，分别是9、42、7、35类

我在这里声明一下有关商标的事宜。
1、在中国我是第一个制作基于openpnp的贴片机并用于商业的。但是从始至终没有使用openpnp作为商标。
2、在中国我注册了openpnp商标，那是因为很多开源项目在中国都被抢注了。例如openMV ，VORON等，并且他们都是因为抢注商标获取了丰厚的利润，为了保护openpnp不会变成被商业所利用的工具，我申请了openpnp商标，并在中国合法取得了正式的商标使用权。这是为了防止有人恶意注册来影响openpnp的发展。
3、截止目前，我没有向任何个人或企业提出诉讼或者维权。在中国的淘宝网站上搜索openpnp，会有很多商品在出售。
4、为了打消大家的疑虑，我愿意将openpnp的商标使用权无偿转让给openpnp的作者和团队，以此来证明我的诚意！
衷心的希望openpnp发展的越来越好！
Alan

Hi, I'm Alan. I am from Hebei Province, China. In June 2020, I made the first openpnp-compatible automatic placement machine. From then on, I knew it was a very promising project and a great one. So I set up a company, built a website, and designed and produced placement machines. After more than two years, we summed up our own shortcomings and achieved certain results. Not long ago, we officially launched the second generation of products.

In China, I applied to register the trademark of openpnp. A total of 4 categories, namely 9, 42, 7, 35 categories

I'm here to make a statement about the trademark.
1. In China, I was the first to make an openpnp-based placement machine and use it commercially. But openpnp was not used as a trademark from beginning to end.
2. I registered the openpnp trademark in China, because many open source projects have been squatted in China. For example, openMV, VORON, etc., and they all gained huge profits because of trademark squatting. In order to protect openpnp from becoming a tool used by businesses, I applied for the openpnp trademark and legally obtained the official trademark use in China right. This is to prevent malicious registration to affect the development of openpnp.
3. So far, I have not filed a lawsuit or defended my rights against any individual or company. Search openpnp on China's Taobao site, there will be a lot of items for sale.
4. In order to dispel everyone's doubts, I am willing to transfer the right to use the openpnp trademark to the author and team of openpnp free of charge to prove my sincerity!
I sincerely hope that openpnp will develop better and better!
Alan

This statement is translated by Google, everything is subject to Chinese meaning!

vonnieda, 2022-12-07:
Thank you Alan. Your statement above is enough for me. I consider this matter resolved and I thank you for your kindness and passion for OpenPnP. 

Please let me know your new domain name when you decide - I will add it to https://openpnp.org/hardware/ under whichever category you prefer.

In addition, I'll note that I'm currently rewriting the openpnp.org site to be Markdown based so that it can be more easily managed in the Github repo. When that is finished, perhaps we could work on translating it to Chinese, and other languages. Then openpnp.cn could point to the Chinese version.

Alan-0131, 2022-12-09:
Thanks jason, this is a big surprise.

I will fully cooperate with you on openpnp.cn.

I'm sorry to reply you so late, because my family has settled in the ICU, so I need to deal with it.

In addition, I want to add the Chinese part of openpnp multilingual. I also made an announcement before. I don't know whether it is appropriate to do so, so I will apply with you here. If it is not appropriate, I will cancel it.
thank！
alan

[image]

Dear friends who support Alan, because the official development team of openpnp has now modified the multilingual system to UTF-8, so that the openpnp multilingual system can better support Chinese. This is exciting news. Since the official development team comes from all countries, Chinese culture needs the joint efforts of our people. As for the open source team, it is different from the commercial team, so we can't just put forward requirements and opinions. We need to do something for the open source organization to build a good ecosystem. So I hope that competent friends can contribute to the Chinese culture of openpnp, provide Chinese translation work according to the operating rules of openpnp, submit it to github, and finally merge it into the distribution version officially by openpnp. I solemnly promise! A bonus of RMB 1000 will be given to each of the top five contributors ranked by github for Chinese language workload. The commitment will be publicly fulfilled when the project reaches 30% of the total workload. We jointly support the healthy development of openpnp and benefit every openpnp user.

vonnieda, 2023-01-15:
Hi @Alan-0131, http://openpnp.cn should work with the OpenPnP site now. https://openpnp.cn is giving a security error currently - I'm working with the host to find out why. 

Please let me know if you have a new domain name you'd like me to add to the Hardware page.

vonnieda, 2023-01-15:
Note: I changed the title from Openpnp trademark squatted to better reflect the positive outcome :)

Alan-0131, 2023-01-17:
thanks a lot jason
Yes, openpnp.cn can be accessed normally, and I will fully cooperate in the follow-up work related to domain names and Chinese culture
Thank you for your help in giving me the link, my new domain name is www.alan0131.com
Thank you, I will fully cooperate with you, and hope that more people can experience the charm of openpnp! Hope openpnp is getting stronger and stronger!

vonnieda, 2023-03-13:
Hi Alan, can you give me an English and/or Chinese description of how you would like the link to appear on https://openpnp.org/hardware/?

Here is a template:

> http://www.alan0131.com/ sells OpenPnP compatible hardware in China, and provides a community resource for Chinese OpenPnP users

Just write it exactly as you'd like it to appear on the site.

I'm going to start working on posting the bug bounties today - so look for an announcement on that soon.

Thanks,
Jason