openpnp/openpnp #1504: Time out after homing A reoccuring problem?

FlashPackets, 2022-12-18:
Issuing a homing command "G28" from a telnet or serial port will generate immediate success with no delay for subsequent commands such as G0 X100 Y100 for instance.  However, a home command issued from the graphical interface homes the machine followed by a lockout lasting many seconds. Eventually a message box will appear with "Timed out waiting for home to complete"
Even more troubling is that the log clearly shows a successful implementation of a small move immediately following G28 to eliminate the possibility that the homing switches are still active as was suggested by another person who experienced the same problem.

In the very early days of my thrashing around the code a homing operation would turn the house black. Now the house stays yellow.

I have now way to know that it is of any significance but the scripts subdirectory of .openpnp2 is empty which seems odd. Do I need to reinstall everything?

[OpenPnP.log](https://github.com/openpnp/openpnp/files/10253187/OpenPnP.log)

FlashPackets, 2022-12-24:
My bad! The problem was resolved by simply placing an M114 command at the end of the HOME_COMMAND. As in
G28
M400
G0 X5 Y5 Z5 A10
M114
Other abbreviated sequences are likely to work just as well.

closed 2022-12-24 by FlashPackets

markmaker, 2022-12-24:
Sorry, I must have overlooked this.

Adding this should not be necessary, it points to a missing command configuration.

Are you using **Issues & Solutions**? It should propose all the right command and other stuff. ic also recommend using the latest test version.

https://github.com/openpnp/openpnp/wiki/Issues-and-Solutions

https://openpnp.org/test-downloads/

_Mark