openpnp/openpnp #1526: Request to edit the Wiki

markmaker, 2023-02-22:
Please leave requests to edit the Wiki here:

aaroneiche, 2023-03-03:
Hello - I'd like to edit the Wiki. Specifically I would like to edit the page on Marlin as it is out of date, and some of the links are broken.

vonnieda, 2023-03-09:
@aaroneiche Done! Thank you for your willingness to help!

Vultik, 2023-03-17:
Is it plan to migrate the Github wiki to real documentation using a tool like https://www.mkdocs.org/ or https://www.sphinx-doc.org/en/master/ ?
The default wiki is hard to navigate between pages and the search is not really great.
For beginners (like me), it will be very beneficial to have it.

Jnesselr, 2023-03-27:
May I and @sphawes have wiki edit access?

vonnieda, 2023-03-27:
> May I and @sphawes have wiki edit access?

Done and done!

janm012012, 2023-05-04:
Hi all!

I'd like to update the wiki with the new Altium importer. Would anyone please give me the required permissions? Thank You!

Jan

vonnieda, 2023-05-05:
> Hi all!
> 
> I'd like to update the wiki with the new Altium importer. Would anyone please give me the required permissions? Thank You!
> 
> Jan

invitation sent!

jkuusama, 2024-02-01:
Re page https://github.com/openpnp/openpnp/wiki/LitePlacer: LitePlacer does have bottom vision.

vonnieda, 2024-02-01:
Hi Juha, I hope you are well! I've sent you an invitation to be able to edit the Wiki. Looks like maybe @markmaker already took care of the issue (?) but regardless, you should be able to edit now.

jsphpl, 2024-03-24:
Requesting write access to add Neoden 4 Linux instructions to the wiki

vonnieda, 2024-03-25:
Invite sent @jsphpl - thanks for contributing!

Jason

hydra, 2024-06-01:
Hi, I would also like to help out, initially by updating the SWD pinout and mentioning that 'VDD' is 3.3V.

tobydickenson, 2024-10-03:
I would like to edit the wiki please

vonnieda, 2024-10-08:
Invites sent @hydra and @tobydickenson. Thanks for your contributions, and sorry for the delay. Just back from travel :)

duncanellison, 2024-10-29:
Hi - I'd like permission to edit the Wiki, probably not a lot, but I'd like to document small points of clarification as I find them.  Specifically I would like to add some comments to the KiuCad importer.

vonnieda, 2024-11-01:
Invite sent @duncanellison, thank you!

tmandys, 2025-02-11:
Hi, I would like to document my pull request feature in wiki. .Tomas

vonnieda, 2025-03-02:
Hi folks, I am currently working through improvements to permissions for the wiki. I'll ping you here once I have your access ready. Sorry for the long wait.

mxxo, 2025-07-03:
Hi, I'd like to update the `Scripting` page to document the `Job.Placement.Starting` and `Job.Placement.BeforeAssembly` events.

scottlawsonbc, 2026-01-01:
I'd like to edit the wiki to update the documentation for the gcodedriver which has diverged from the source code
https://github.com/openpnp/openpnp/blob/39973975b6de820e1e67e7ae43fea0bb0da01b9b/src/main/java/org/openpnp/machine/reference/driver/GcodeDriver.java#L91
I'd like to update the command reference documentation listing the commands and regexes, it is currently missing some and hasn't been updated since 2022

lichtdompteur, 2026-01-20:
I would like to fix some typos in the wiki

vonnieda, 2026-01-22:
Thanks @Jorropo and @tobydickenson, these are both good ideas and seem pretty easy to implement. I'll check them out.

arthurwolf, 2026-10-13:
Hey, can somebody edit the wiki or give me access to do so, it's a pretty simple change, just for https://github.com/openpnp/openpnp/wiki/Motion-Controllers#smoothie it currently says that v2 is not yet available, and it's been available for a couple of years now (after **a lot** of sweat and tears...), so objectively inaccurate info, and were starting to have time to put a lot more effort back into the documentation and firmware, and part of that is making sure information online about it is correct, and this is one thing i found that would benefit from an update.

I can give exact text as a proposal if it's helpful, but this is a pretty simple fix.

I think it'd also make sense to either list both the v1 and v2 boards in the list of boards, or at least replace the v1 board with the v2 board.

Tell me if you need anything from me and if I can do anything to help.