openpnp/openpnp #1537: Documentation Improvements

vonnieda, 2023-03-20:
Is it plan to migrate the Github wiki to real documentation using a tool like https://www.mkdocs.org/ or https://www.sphinx-doc.org/en/master/ ?
The default wiki is hard to navigate between pages and the search is not really great.
For beginners (like me), it will be very beneficial to have it.

_Originally posted by @Vultik in https://github.com/openpnp/openpnp/issues/1526#issuecomment-1474229547_

vonnieda, 2023-03-20:
I created this issue to discuss making some improvements to OpenPnP's documentation. As @Vultik mentioned, the Wiki is difficult to use.

Improving the documentation is something I'm interested in spending some time on, so I thought I'd collect some ideas here.

Here are a few things I have in mind:

- Consider porting to a true documentation system, as @Vultik mentioned.
- Consolidate duplicated information.
- Contextual help in OpenPnP, linking to a local copy of the manual.
  - More tooltips or hover info.
  - If adding a tooltip, can also add a link to the details in the manual.
  - A local copy is great for operators; machines may not be connected to network.

matmair, 2023-03-21:
Several projects I contribute to use Mkdocs Material: https://squidfunk.github.io/mkdocs-material/. Very nice feature set and the content is easy to translated into other systems (like sphinx).
@vonnieda would you want to just move over the content of the wiki or rework it too? I'd be open to help with the technical/CI side - my knowledge of OpenPnP is limited so I can not really help with content.

ian-arkver, 2023-03-21:
There are 179 links from OpenPnP files to the github wiki which will need changed if that is retired or becomes stale.

grep -r openpnp/wiki src *.md | wc -l
179

markmaker, 2023-03-22:
The only things I want to say is that it should be Open Source and somehow hooked into the git/Github versioning. It seems MkDocs fits that bill.

(I know Github is also not Open Source, but then again the dependency is already there with the whole repo)

_Mark

matmair, 2023-03-25:
I think the best would be to move the content into the main repo, then versioning is coupled and you have it available portably.

@markmaker @vonnieda Besides the current sidebar, is there any structure in the wiki that should be replicated? I can not see any organisation other than the sidebar in the current wiki.

vonnieda, 2023-03-28:
MkDocs sounds great to me. I agree - keeping the docs in the main repo would be a huge improvement. I've always been frustrated that the Wiki was so decoupled. 

@matmair The only other structure I can think of is just that of the permalinks for the existing entries. Several are referenced in the software - so those either need to be maintained or updated.

matmair, 2023-03-30:
I am working on a POC that preserves the historical edits [here](https://github.com/matmair/openpnp-wiki/tree/add-mkdocs). Still tinkering with the structure and CI hooks. I will create a PR once I am somewhat happy with it.

matmair, 2023-04-02:
@vonnieda @markmaker I am getting ready to open a PR.
1) Where in the repo should the docs be positioned? In `/docs` or somewhere else?
2) Do you want to keep the editing history for the wiki pages? I set it up to keep them but it is 2.2k commits

vonnieda, 2023-04-03:
@matmair Thank you for your work on this! Answers:

1. /docs or whatever your preference that isn't already taken sounds fine. I defer to you.
2. I think we can do without the history. I'll archive the current Wiki somewhere people can grab it. Only suggestion would be that if you have the option of maybe including the last 100 commits or something, that would be nice. Totally not required though.

Is there a place to view the rendered version of your work so far, or would I clone and build myself?

Thanks,
Jason

matmair, 2023-04-03:
Thank you for the answers. I will try to finish cleaning up the markdown style and links (a lot of things are linked to the full wiki urls, not the documents internally) in the next 1-2 weeks and provide a PR with all necessary the CI and a shortened history.
You can view an experimental build [here](https://openpnp-wiki.readthedocs.io/en/latest/Sidebar/Development%20Resources/Your-First-Pull-Request.html). That is the raw content of the wiki, I still need to fix a bunch of things.

vonnieda, 2023-04-03:
That looks excellent so far!

alexanderwiller, 2025-03-22:
Searching for OpenPnP documentation using Google, I noticed that by now quite a few results do not point to the GitHub wiki, but to https://openpnp-wiki.readthedocs.io/.

Until I searched for this issue, as well as for #1543, https://github.com/openpnp/openpnp/issues/1526#issuecomment-1474229547, I did not understand how that page relates to the GitHub wiki. **This creates a great deal of confusion.**

I am a strong advocate for keeping the documentation with the code, and to use a docs generator like discussed in this and the linked issues.

Even before getting confused by the readthedocs.io version of the wiki, we (wearing the _OpenPnP user_ hat) struggled quite a bit to locate information in the GitHub wiki, as some pages are not visible in the index and have to be found by other means.

In the past, when I submitted a PR, for me the current solution also created a noticeable disconnect between the code changes and the documentation changes, requiring an "out of band" workflow for the documentation update.

**Why am I writing this? I think that it is high time for this topic to be addressed with a relatively high priority. I believe good documentation is critical to the long-term success of a complex OSS project such as OpenPnP. The alignment of Wiki and Code will not get better over time.**

---

I want to answer some of @markmaker's questions from https://github.com/openpnp/openpnp/pull/1543#issuecomment-1501864522 as a minimal contribution to restart the discussion:

**How will we actually author the documents, exactly?**

In general, with a plain text editor of your choice, potentially with extension for preview rendering and/or WYSIWYG editing.

The details depend on the chosen documentation generator, for which I see two main options: Generators that use markdown (like mkdocs) and generators that use asciidoc (which mostly means Antora: https://antora.org/)

I can only speak for the Asciidoc/Antora pair, as I have used that extensively in practice.

For asciidoc, there is extensive asciidoc language and Antora support in at least two major IDEs:
* vscode: https://marketplace.visualstudio.com/items?itemName=asciidoctor.asciidoctor-vscode (I use it and it works)
* IntelliJ: https://intellij-asciidoc-plugin.ahus1.de/docs/users-guide/features/advanced/antora.html (Have never tried it)

Both the vscode and the IntelliJ plugins include a preview renderer, as well as support for resolving Antora-type links between files in the local preview, also resolution of images and other assets.

For further background on tooling check out https://docs.asciidoctor.org/asciidoctor/latest/tooling/. There's also a standalone editor called https://asciidocfx.com/.

See at the very end of this comment for some further info what asciidoc is even about.

To summarize, document authoring happens in your local IDE or text/WYSIWYG editor.

**For us developers, I assume we can now make the document (markup) editing part of the normal PRs, right?**

That is correct and one of the key benefits of this flow.

**What about non-developers, will there be an online editor (similar to the Wiki editor on Github) available?**

Not sure, with these solutions (any document generator) I'd say it's not typical to have a web-based editor right inside the documentation. There are some solutions (https://github.com/EddyLuten/mkdocs-live-edit-plugin) but I do not know if they are a good idea.

One could argue that the GitHub Codespaces or other online editors provided by typical git forges are an "online editor", but probably not in the sense of this question.

So, generally, documentation edits would require a PR (or at least an issue with a change set that is transferred to the docs). Maybe it makes sense to have a community wiki that is editable on the web, but that is over my head for now. I think considering if "official" documentation and a community wiki are two different things is worthwhile.

**What about pictures? Where do we put them? Is this all manual? (Github hat automatic upload with copy&paste or drag).**

For asciidoc, Antora automatically locates resources from a designated directory if they are referenced by an `image:example.png[This is an example]` tag. For Markdown, I am sure that similar solutions exist.

I know that the vscode extension for asciidoc supports `Ctrl + Alt + V` to paste images right into the document, putting the file in the correct resource folder. For markdown, I expect a similar functionality to exist.

By default, the images are named with a timestamp when using this paste feature, but as all changes go through a MR workflow, it is encouraged to actually give image files meaningful names - some extra upfront effort, but it comes in very handy when images need to be re-used across pages.

**What happens if we keep editing the old Github Wiki now? Will theses changes be lost?**

I think that generally, no, no data loss will occur - assuming a proper migration plan is created.

Such a migration plan would consist of putting all the infrastructure into place to generate docs right from the repository, then taking a content snapshot of the wiki and making it read-only after that.

From this point on, all edits need to be performed on the repo directly. To keep it practical, I would expect that the content structure is migrated 1:1 at first (or with minor alterations to be compatible with the chosen documentation generator), and then cleaned up progressively to make use of the new documentation gene [... cut ...] esigned for authoring of complex multi-page technical documentation from the start, while for markdown, this is more of an afterthought.

Just a few links on this:

* https://docs.asciidoctor.org/asciidoc/latest/asciidoc-vs-markdown/
* https://docs.asciidoctor.org/asciidoc/latest/syntax-quick-reference/
* https://docs.asciidoctor.org/asciidoctor/latest/migrate/markdown/ - I expect that a migration is quite viable, which can be proven by the "test migration" described above.

Hope this helps
Alex

alexanderwiller, 2025-03-23:
Just a few more remarks on the current state of "documentation visibility":

Only coincidentally yesterday I discovered how all pages can be listed. I've been consulting the OpenPnP docs since 2021 or so, and I never realized there was a "full" navigation. I know two other users of this documentation which I would call "experienced in reading technical documentation", neither of them were aware of this full index either.

In addition, when expanding the navigation to show all sections, it becomes pretty much unmanageable, I couldn't even fully capture it in my screenshot.

<details>

<summary>Expand section for screenshots of current navigation presentation</summary>

[image]
[image]
[image]

</details>