openpnp/openpnp #1538: Won't start on Windows after #1534

vonnieda, 2023-03-21:
Roman reports OpenPnP won't start after update: in https://groups.google.com/d/msgid/openpnp/bc21e865-c9c7-4162-a39b-e37b97d3a9b1n%40googlegroups.com?utm_medium=email&utm_source=footer

vonnieda, 2023-03-21:
Looks like they're running Windows 7. Quite probable that the new openpnp-capture was built too new.

tonyluken, 2023-03-26:
I suspect this is more than a Windows 7 issue, people are having problems under Windows 10 and 11 as well.  See discussion [here](https://groups.google.com/g/openpnp/c/0VC11m_59dM)

vonnieda, 2023-03-27:
Confirmed that the new openpnp-capture binaries are broken, and am working on fixing them.

vonnieda, 2023-03-28:
This should be fixed now, in the test branch. Please try the latest: https://openpnp.org/test-downloads/

Issue was due to new openpnp-capture version being built with MinGW instead of Visual Studio. It is now being built with Visual Studio 2022. There is still a risk that it might not work on much older versions of Windows, but you should be able to install a support package.

Please let me know via this thread or otherwise if you still experience startup issues.

Thanks!
Jason

tonyluken, 2023-04-02:
@vonnieda,

I'm not sure this is related but I'm now seeing this stack trace when I try to exit from OpenPnP:

2023-04-02 13:31:59.880 MainFrame INFO: Shutting down...
2023-04-02 13:32:00.114 ReferenceMachine DEBUG: setEnabled(false)
2023-04-02 13:32:00.115 ReferenceMachine INFO: setHomed(false)
2023-04-02 13:32:00.570 AbstractBroadcastingCamera TRACE: Camera Down Camera thread 55 bye-bye.
2023-04-02 13:32:00.575 AbstractBroadcastingCamera TRACE: Camera Up Camera thread 57 bye-bye.
2023-04-02 13:32:00.881 SystemLogger ERROR: Exception in thread "AWT-EventQueue-0" java.lang.Error: Invalid memory access

2023-04-02 13:32:00.891 SystemLogger ERROR: 	at com.sun.jna.Native.invokeInt(Native Method)

2023-04-02 13:32:00.891 SystemLogger ERROR: 	at com.sun.jna.Function.invoke(Function.java:426)

2023-04-02 13:32:00.891 SystemLogger ERROR: 	at com.sun.jna.Function.invoke(Function.java:361)

2023-04-02 13:32:00.891 SystemLogger ERROR: 	at com.sun.jna.Library$Handler.invoke(Library.java:265)

2023-04-02 13:32:00.891 SystemLogger ERROR: 	at com.sun.proxy.$Proxy21.Cap_closeStream(Unknown Source)

2023-04-02 13:32:00.891 SystemLogger ERROR: 	at org.openpnp.capture.CaptureStream.close(CaptureStream.java:26)

2023-04-02 13:32:00.891 SystemLogger ERROR: 	at org.openpnp.machine.reference.camera.OpenPnpCaptureCamera.close(OpenPnpCaptureCamera.java:299)

2023-04-02 13:32:00.891 SystemLogger ERROR: 	at org.openpnp.machine.reference.ReferenceMachine.close(ReferenceMachine.java:554)

2023-04-02 13:32:00.891 SystemLogger ERROR: 	at org.openpnp.gui.MainFrame.quit(MainFrame.java:1089)

2023-04-02 13:32:00.891 SystemLogger ERROR: 	at org.openpnp.gui.MainFrame$21.windowClosing(MainFrame.java:296)

2023-04-02 13:32:00.891 SystemLogger ERROR: 	at java.desktop/java.awt.AWTEventMulticaster.windowClosing(AWTEventMulticaster.java:357)

2023-04-02 13:32:00.891 SystemLogger ERROR: 	at java.desktop/java.awt.Window.processWindowEvent(Window.java:2078)

2023-04-02 13:32:00.892 SystemLogger ERROR: 	at java.desktop/javax.swing.JFrame.processWindowEvent(JFrame.java:298)

2023-04-02 13:32:00.892 SystemLogger ERROR: 	at java.desktop/java.awt.Window.processEvent(Window.java:2037)

2023-04-02 13:32:00.892 SystemLogger ERROR: 	at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5008)

2023-04-02 13:32:00.892 SystemLogger ERROR: 	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2321)

2023-04-02 13:32:00.892 SystemLogger ERROR: 	at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2772)

2023-04-02 13:32:00.892 SystemLogger ERROR: 	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4840)

2023-04-02 13:32:00.892 SystemLogger ERROR: 	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:772)

2023-04-02 13:32:00.892 SystemLogger ERROR: 	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)

2023-04-02 13:32:00.892 SystemLogger ERROR: 	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)

2023-04-02 13:32:00.892 SystemLogger ERROR: 	at java.base/java.security.AccessController.doPrivileged(Native Method)

2023-04-02 13:32:00.892 SystemLogger ERROR: 	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)

2023-04-02 13:32:00.892 SystemLogger ERROR: 	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:95)

2023-04-02 13:32:00.893 SystemLogger ERROR: 	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:745)

2023-04-02 13:32:00.893 SystemLogger ERROR: 	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:743)

2023-04-02 13:32:00.893 SystemLogger ERROR: 	at java.base/java.security.AccessController.doPrivileged(Native Method)

2023-04-02 13:32:00.893 SystemLogger ERROR: 	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)

2023-04-02 13:32:00.893 SystemLogger ERROR: 	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:742)

2023-04-02 13:32:00.893 SystemLogger ERROR: 	at org.openpnp.gui.MainFrame$22.dispatchEvent(MainFrame.java:556)

2023-04-02 13:32:00.893 SystemLogger ERROR: 	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)

2023-04-02 13:32:00.893 SystemLogger ERROR: 	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)

2023-04-02 13:32:00.893 SystemLogger ERROR: 	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)

2023-04-02 13:32:00.893 SystemLogger ERROR: 	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)

2023-04-02 13:32:00.893 SystemLogger ERROR: 	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)

2023-04-02 13:32:00.893 SystemLogger ERROR: 	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)

lags, 2023-04-02:
Seems to relate to `openpnp-capture`
Maybe [here](https://github.com/openpnp/openpnp-capture-java/blob/main/src/main/java/org/openpnp/capture/library/OpenpnpCaptureLibrary.java#L19) could be fixed changed to:

`public static final OpenpnpCaptureLibrary INSTANCE = Native.load(OpenpnpCaptureLibrary.JNA_LIBRARY_NAME, OpenpnpCaptureLibrary.class);`

Unfortunately I can't test it now.
Hoping I haven't said nonsense.

vonnieda, 2023-04-03:
Alright, I made that change and pushed it for testing. I'm working on a couple other issues in openpnp-capture, so I'll be updating everything across the board to v0.0.26 here briefly.

vonnieda, 2023-04-03:
OpenPnP has been updated to v0.0.26 but it may take a few hours before everything becomes available. You can watch for when this goes green, or at lest mostly green: https://github.com/openpnp/openpnp/actions/runs/4602385838

vonnieda, 2023-04-04:
Temporarily rolling this back while the artifact gets released.

markmaker, 2023-04-11:
Hi @vonnieda ,

Not sure about the status here. I currently can't use OpenPnP on Windows 11. 

Using Eclipse, it runs until I want to connect to a camera, then it crashes silently, including the Java runtime. 

```xml
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
<System>
  <Provider Name="Application Error" Guid="{a0e9b465-b939-57d7-b27d-95d8e925ff57}" /> 
  <EventID>1000</EventID> 
  <Version>0</Version> 
  <Level>2</Level> 
  <Task>100</Task> 
  <Opcode>0</Opcode> 
  <Keywords>0x8000000000000000</Keywords> 
  <TimeCreated SystemTime="2023-04-11T07:17:10.6795829Z" /> 
  <EventRecordID>23180</EventRecordID> 
  <Correlation /> 
  <Execution ProcessID="11980" ThreadID="22092" /> 
  <Channel>Application</Channel> 
  <Computer>vulture</Computer> 
  <Security UserID="S-1-5-21-2799649978-392620158-41799015-1001" /> 
  </System>
  <EventData>
  <Data Name="AppName">java.exe</Data> 
  <Data Name="AppVersion">11.0.18.0</Data> 
  <Data Name="AppTimeStamp">63c8174e</Data> 
  <Data Name="ModuleName">ucrtbase.dll</Data> 
  <Data Name="ModuleVersion">10.0.22621.608</Data> 
  <Data Name="ModuleTimeStamp">f5fc15a3</Data> 
  <Data Name="ExceptionCode">c0000409</Data> 
  <Data Name="FaultingOffset">000000000007f61e</Data> 
  <Data Name="ProcessId">0x4484</Data> 
  <Data Name="ProcessCreationTime">0x1d96c45944dda21</Data> 
  <Data Name="AppPath">C:\Program Files\Microsoft\jdk-11.0.18.10-hotspot\bin\java.exe</Data> 
  <Data Name="ModulePath">C:\Windows\System32\ucrtbase.dll</Data> 
  <Data Name="IntegratorReportId">f8b732e1-e998-4ead-a3c7-5ef310cb53ae</Data> 
  <Data Name="PackageFullName" /> 
  <Data Name="PackageRelativeAppId" /> 
  </EventData>
  </Event>
```
Because I'm currently using a Java runtime built by Microsoft, and it crashes in the `ucrtbase.dll` which strangely is a .NET file, I first suspected Microsoft as the culprit, so I also tested with OpenPnP installer's packaged runtime (`test` version).

Using the installer it fails immediately with the following exception (excerpts):

```
2023-04-11 09:21:40.410 SystemLogger ERROR: java.lang.Exception: Error while reading machine.xml (Unable to load library 'openpnp-capture':
...
2023-04-11 09:21:40.415 SystemLogger ERROR: Caused by: java.lang.UnsatisfiedLinkError: Unable to load library 'openpnp-capture':
Native library (win32-x86/openpnp-capture.dll) not found in resource path (C:\Program Files (x86)\openpnp\.install4j\i4jruntime.jar;C:\Program Files (x86)\openpnp\.\openpnp-gui-0.0.1-alpha-SNAPSHOT.jar;...)

2023-04-11 09:21:40.415 SystemLogger ERROR: 	at com.sun.jna.NativeLibrary.loadLibrary(NativeLibrary.java:301)

2023-04-11 09:21:40.415 SystemLogger ERROR: 	at com.sun.jna.NativeLibrary.getInstance(NativeLibrary.java:461)

2023-04-11 09:21:40.415 SystemLogger ERROR: 	at com.sun.jna.NativeLibrary.getInstance(NativeLibrary.java:403)

2023-04-11 09:21:40.415 SystemLogger ERROR: 	at org.openpnp.capture.library.OpenpnpCaptureLibrary.<clinit>(OpenpnpCaptureLibrary.java:18)

2023-04-11 09:21:40.415 SystemLogger ERROR: 	at org.openpnp.capture.OpenPnpCapture.<init>(OpenPnpCapture.java:15)

2023-04-11 09:21:40.415 SystemLogger ERROR: 	at org.openpnp.machine.reference.camera.OpenPnpCaptureCamera.<init>(OpenPnpCaptureCamera.java:42)

```

No clue if it is relevant, but I found it strange that it installs OpenPnP into the 32bit programs folder on a 64bit system.

Any idea? Thanks.

_Mark

markmaker, 2023-04-12:
Hi @vonnieda,

Tried the `features/upgrade_openpnp_capture` branch locally. Unfortunately, it does not change the outcome. Still crashes silently, same log entry in the Event Viewer.

This time I also alternatively tried with Java 20. Same outcome. 

I wish I could help, but I'm completely clueless regarding JNI and toolchains.

_Mark

markmaker, 2023-04-12:
See https://github.com/openpnp/openpnp-capture/pull/65

vonnieda, 2023-04-17:
This has been merged and users are reporting it's working.

closed 2023-04-17 by vonnieda