openpnp/openpnp #1545: Calibrate precise camera - nozzle offsets error

Dabihu, 2023-04-15:
## Describe the Issue
Version: 2023-03-15_00-30-21.460d8aa
Windows 10
Tab: Issues & Solutions
Subject: ReferenceNozzle N1/N2
Issue:  Calibrate precise camera - nozzle N1/N2 offsets

The algorithm works from the machine.xml file, not from the current configuration. For precise calibration it must do 3 times, every time with a restarted program.

## Steps to Reproduce
1. Make an offset error in N1 or N2 (I had X=0.22 mm offset error)
2. Close program
3. Start OpenPnP
4. Home
5. Reopen this issue and do the calibration
6. Close program
7. Repeat step 3..6  3 times

If you don't close the program, the calibration calculate the same correction every time without progress.

## Expected Result
1. Do a precise calibration from actual configuration, without need to restart the program.
2. Do a precise calibration after the first run. Not needed to do it 3 times.

markmaker, 2023-04-15:
Hi @Dabihu,

this might be a misunderstanding:

1. **Issues & Solutions** stores the old offsets like an "Undo"-Memory. 

3. If you press **Reopen** immediately after the calibration, it will _restore_ the old offsets. See the tooltip on **Reopen**:

    [image]
 
5. If you want an _incremental_ calibration, there is no need to exit OpenPnP, just press **Find Issues & Solutions** to reset the "Undo"-Memory.
 
7. Check **Include Solved?** for the solution to reappear.
 
9. Press **Reopen**.

11. Press **Accept** to redo the calibration. It should now only make a tiny adjustment (representing the residual error in your machine-camera system).
 

Please report back if this helps, or not.

_Mark

Dabihu, 2023-04-15:
Hi Mark,

Sorry, this is my fault!
I didn't know there was a temporary undo memory in Issues & Solutions.
I tried to find where the old value stored, but it was not clear, that a "Find Issues & Solutions" clear this memory.

The other problem was solved somehow. In this case, the 0.22 mm offset was reduced to 0.005 mm in one step. I don't understand what happened before.

Thank You Mark!

markmaker, 2023-04-15:
So do you think this can be closed?

Dabihu, 2023-04-15:
Yes, I tested in the past hours and can not found the root of precision error. +- 0.1 mm sometimes. But don't think it's a software error.
Thank you!

closed 2023-04-15 by Dabihu

markmaker, 2023-04-16:
> Yes, I tested in the past hours and can not found the root of precision error. +- 0.1 mm sometimes.

Best go to the [OpenPnP discussion group](http://groups.google.com/group/openpnp), and open a new conversation there. Maybe we can help.

_Mark