openpnp/openpnp #1551: [Feature Request] add GenICam support to openpnp

GridWork2, 2023-04-29:
# Feature Request
## Describe the Feature
I would like to see Gen<I>Cam support added to the available camera drivers for OpenPNP. According to emva.org, Gen<I>Cam is a vision standard for machine vision cameras, both USB and GigE.

## How Will It Look?
I can't say exactly what the final implementation would look like, I could envision just another item for GenICam camera in addition to the OpenCV or OpenPNPcamera instances. Selecting that would allow assigning either usb and associated properties or ethernet and appropriate address. In addition all of the traditional image/camera properties would be accessible, similar to existing camera drivers. 

## Why Do We Need It?
1. What problem does it solve?
There are a large number of industrial cameras available on the surplus/used market, many of these are current enough to support the GenICam standard. In addition, the number of possible cameras available for a machine build grows exponentially, expanding from the typical webcam or ELP style board camera. In theory it should also make a Linux deployment go a little smoother as many of these camera manufacturers support Linux for machine vision.

Many of the industrial cameras are also substantially higher performance, incorporating USB3 or GigE. Larger sensors and thus larger fields of view are also more available. GenICam also can contain the ability for ethernet connections, which are substantially more robust for reliability (vs. USB).

2. Is it useful for everyone, or does it only solve a problem for your machine?
I assert that this feature would be useful for a great number of users, particularly anyone doing a conversion of an older commercial machine. Being able to re-use many of the potential surplus cameras available also fits very nicely in reducing e-waste and re-purposing it.

## Provide Examples
Is this something that exists in other pick and place software? Provide examples and references if possible.
I am unsure if it exists in alternative machines, however I would imaging commercial/professional machines would have something similar.

https://www.emva.org/standards-technology/genicam/genicam-downloads/

markmaker, 2023-04-30:
Hi @GridWork2,

Just to be sure: is this just a plain Feature Request, or is also an offer/announcement that you would like to develop and contribute such a thing?

I've quickly searched for products supporting the API, and only found very expensive cameras. Do you know some that are affordable? Or do you have/know a machine commonly retro-fitted for OpenPnP that has one?

_Mark

GridWork2, 2023-04-30:
I would love to contribute, but my programming experience is much more limited to microcontrollers. 

Some of the cameras available are definitely more expensive (when new), however there are some starting to appear in the 200 USD range (new). One of the main reasons I think that GenICam might be useful is in the performance of these cameras is much better. 

Fiddling with an ELP style board camera, the video stream either can be high frame rate with MJPEG compression or raw data at low frame rate. This appears to be a limit of the UVC USB2 specification (I might be wrong about this). The ELP cameras also appear to generally be color only. A monochrome camera improves resolution somewhat and also opens up optical filtering capabilities.

I do see industrial machines frequently show up on the market. Quad, Zevatech/Juki, Samsung and MyData seem to be common ones. Most of the time these are retired from build houses due to inability to place fine parts. This seems mostly to be due to the software on them. Most are 15 to 30 years old, so a good number of improvements have been made. These machines make good candidates for a DIY rebuild for someone whos serious, as the motion systems are usually still in good shape. I have two Zevatech 560's, one is in use, the other is undergoing an overhaul to OpenPNP control.

I'd love to help in bringing this feature to life as I think quite a few people would benefit.

markmaker, 2023-04-30:
> Some of the cameras available are definitely more expensive (when new), however there are some starting to appear in the 200 USD range (new).

Cool, links, please.

>  This appears to be a limit of the UVC USB2 specification

Guess so.

>  A monochrome camera improves resolution somewhat and also opens up optical filtering capabilities.

I beg to differ with the latter. Filtering is IMHO better done in software, i.e. it can then be calibrated to the machine/colors/etc. Available color information can be very useful for Computer Vision. There are already well-established routine applications in OpenPnP, including software-based filtering calibration:

https://github.com/openpnp/openpnp/wiki/Nozzle-Tip-Background-Calibration

[image]

Similar applications for the top camera:

https://github.com/openpnp/openpnp/wiki/BlindsFeeder

[image]

_Mark

ian-arkver, 2023-04-30:
It is possible that [Aravis](https://github.com/AravisProject/aravis) already includes a gstreamer plugin for GenICam capture which would then work with GStreamerCamera. If this is so, and there's interest, I can take another look at getting GStreamerCamera merged.

markmaker, 2023-05-01:
This would be ideal, as I think (hope) G-streamer API would also improve the bad Linux support, generally. 

However, I hope the important goodies of the `OpenPnpCaptureCamera` could be refactored into a common base class. 

_Mark

ian-arkver, 2023-05-01:
Hi Mark,

Yeah, not really. There aren't any easy consistent connections between the sliders & dropdowns of the OpenPnPCaptureCamera's nice settings and generic GStreamer pipelines.

The idea with GStreamerCamera was that it was extremely bare-bones and left all the camera settings to be done outside of OpenPnP in whatever way was needed for that camera/pipeline. This gives the most flexibility, but does shift the onus somewhat onto the user to set things up themselves.

For example, the pipelines I use are actually just RTP/UDP socket listeners piped into HW accelerated H265 decoders. The cameras are on a different machine. There are no controls at the OpenPnP side whatsoever.

I think for users that like the sliders and settings of OpenPnPCaptureCamera they'd probably be better sticking with that.

Regards,
Ian

ian-arkver, 2023-05-01:
[Take a look at the commit](https://github.com/ian-arkver/openpnp/commit/a8c75a8547212229b98064115b1a6a5ed5f82428) if you like. It's a very thin specialisation of ReferenceCamera.