openpnp/openpnp #1557: Photon Feeder - No PartId Prevents Configuration Save

warasilapm, 2023-05-11:
# Problem Report
## Describe the Issue
When using the photon feeder type with the opulo 8mm feeder, if the feeder does not have a PartId attached to it, the machine configuration will not save. See error below.

## Steps to Reproduce
1. Start OpenPnP, and power machine on.
2. Add manually or by searching in the photon feeder global config tab a photon feeder.
3. Do not select a PartId for the feeder.
4. Attempt to save configuration using the File menu.
5. The machine configuration does not save and generates the error shown above.

## Expected Result
The machine configuration to save.

## Actual Result
The machine configuration does not save successfully. Changes to other areas of the machine configuration are also not affected by this change; it appears any and all changes do not save properly.

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
[image]

I also attempted to create this issue by creating a new ReferenceStripFeeder. However, it would appear that `FIDUCIAL-HOME` is always selected by default. Maybe this is the expected safe state for feeder part IDs?

Jnesselr, 2023-05-14:
This error is in the AbstractFeeder and it's part of the base feeder. I suspect this is because OpenPnP generally considers a feeder as "parts you care about right now" and doesn't usually deal with feeders that aren't there to serve a part (whereas our feeders stay in OpenPnP even if they're off the machine) or may not have an associated part.

@vonnieda @markmaker Any reason why required is true here for the part id?

markmaker, 2023-05-16:
Hi @Jnesselr,

This was defined before my time, but I think a feeder just _always_ needs a part assigned. When you create a new feeder, and no part is given, it defaults to the very first part, here:

https://github.com/openpnp/openpnp/blob/460d8aa4bbfa2a59af1afd2c6e03487d4ba63406/src/main/java/org/openpnp/gui/FeedersPanel.java#L497

Plus the UI does not allow to select a `null` part.

So when you create a `PhotonFeeder` automatically, you need to make sure the same happens, for instance here:

https://github.com/openpnp/openpnp/blob/2897bb97f4bf319fe26e2165a747e1870f6107ec/src/main/java/org/openpnp/machine/photon/PhotonFeeder.java#L204-L206

_Mark

bdebyl, 2023-05-16:
The way I've gotten around this, on my own feeders unrelated to the Photon feeder, is I just have a specific "dummy" part that is not used by any job and marked as such -- can be anything valid really, Fiducials may even work.

warasilapm, 2023-05-17:
`FIDUCIAL-HOME` is what my setup defaults to on all other feeders.

markmaker, 2023-05-17:
>  If I make it optional and make sure there's no crashes, is that acceptable?

That would be very welcome! 💯 😎

It always irked me that this random first part is assigned.

But do not underestimate the needed work. You'd need to ...

1. Check and sanitize each and every call to `Feeder.getPart()` and `Feeder.partId`. 
2. Add the `null` Part selection to the GUI (`ComboBox`, may be handled as empty String), so it is properly displayed, and can also be set.
3. Check where other feeders are created programmatically, and make them default to the `null` part too. 
 
_Mark