openpnp/openpnp #1566: Camera controls don't work on OSX 13.0 (Ventura)

kargeor, 2023-06-07:
# Problem Report
## Describe the Issue
Manual camera controls (exposure, etc) don't work on OSX 13.0 (Ventura). 
Version 2022-12-02 from latest releases.

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. Start OpenPnP
2. Machine Setup.
3. Cameras

## Expected Result
Able to control exposure, brightness, etc.

## Actual Result
All options are disabled.

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
Issue fixed when manually compiling openpnp-capture from source.

markmaker, 2023-06-07:
This should be fixed, but is currently still in testing, therefore, please use the `test` version of OpenPnP and report back here.

https://openpnp.org/test-downloads/

_Mark

markmaker, 2024-10-01:
No news assumed to be good news.

closed 2024-10-01 by markmaker