openpnp/openpnp #1588: cannot resize window, shrinking camera side

tomse, 2023-09-15:
# Read This First
Issues must only be opened once they are confirmed.
Please [visit the discussion group](https://groups.google.com/group/openpnp) first, so bugs or missing features can be separated from other problems.

To submit an issue for OpenPnP you must use this template or it will be deleted. Please fill out either the Problem Report or Feature Request section below and delete the other section.

-----------------------------------------------------------------------

# Problem Report
## Describe the Issue
Cannot resize the windows -> shrinking the camera side

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. Start OpenPnP
2. Move separator slider between config side and camera side towards the right
3. Close OpenPnP
4. Try moving the same slider to the left, enlarging the config side, shrinking the camera side

It seems to save the position of the slider when OpenPnP closes, and sets a lock on that position

## Expected Result
Moving the slider should resize both sides of the slider

## Actual Result
Moving the slider to the left is not possible.. it's "stuck"

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
The more information you can provide about the issue the more likely it is that a developer will be able to respond.

Video of the slider stops trying to move to the left
https://youtu.be/6iu8SAKTY5Q

Whenever possible, please use the Menu -> Help -> Submit Diagnostics feature and include the URL here. Submit Diagnostics makes it easy to include log files, screen capture, debug images, configuration files, etc.

It was not possible to send log files - pastebin error occured with a long url (maybe a bug/api change?)

-----------------------------------------------------------------------

tonyluken, 2023-09-17:
I haven't been able to duplicate the issue with the latest Test branch on Windows 10. What version of OpenPnP are you running and on what OS?

tomse, 2023-09-18:
Running Win 10 22H2

And the latest stable link from the official website (https://s3-us-west-2.amazonaws.com/openpnp/OpenPnP-windows-x64-develop.exe) downloaded 1-2 weeks ago (sorry don't have access to the setup until tomorrow, can't give you a version number atm.

markmaker, 2023-09-18:
It is a known bug, or perhaps "inconsistency", i.e. I guess there is some code in there giving the camera side a certain percentage of the width, once granted.

There is a workaround: you can shrink the whole application window, which will force the camera side smaller. Once you're happy, enlarge the main window again. I guess this workaround is the reason is hasn't been fixed yet, i.e., those knowing the trick no longer complain. :smirk: 

Unfortunately, I'm not at all familiar with Java Swing layouting, and haven't found (and much searched for) the culprit yet. 

Somebody? :100: :1st_place_medal: 

_Mark