openpnp/openpnp #1592: Openpnp on Jetson, custom Opencv

elfarolab, 2023-10-14:
Hello,

I am trying to setup Openpnp on Nvidia Jetson Xavier. I would like to use a custom version of Opencv with Cuda and GPU enabled. I tried to look around and tried several things during the past days but without success.

I already have my custom opencv-481.jar compiled and ready to use. 

How can I use it inside openpnp? Should I rebuild openpnp? How to configure the build process to use my custom jar? Should I modify the openpnp code for to allow calling the external/system library? Any help will be so much appreciated.

I am using openjdk-11 because I understood that 12 doesn't support shared libraries anymore.

Ubuntu 20.04

Thank you so much.

F.

ian-arkver, 2023-10-14:
None of the code in OpenPnP's vision processing uses GpuMat or any of the CUDA accelerated methods. Simply replacing opencv with a cuda enabled version won't achieve any CUDA use or speedup.

elfarolab, 2023-10-14:
I understand.
Yesterday I was thinking about it.

What about enabling Opencv to run on GPU? Do you see any benefits in term of speed?

Thank you so much.

ian-arkver, 2023-10-14:
Sure, yes, I was curious about it a few years ago too. There are other problems than just the OpenCV image processing step though. All the OpenPnP camera capture -> vision pipeline stuff is Java of course, and uses Java memory buffers for the images. To really see the benefit of HW Acceleration you'd need to try to keep the memcpy's down to a minimum and keep the video in NVMM (GPU) space as much as possible. I'm not sure how easy that is within a Java VM.

If you end up copying between a Java VM allocated buffer and an nvidia nvhost allocated NVMM buffer and/or Cuda Device Memory then all the benefit of CUDA acceleration will probably be lost. Sure in a unified memory system like Xavier they're all theoretically the same, but actually mapping everything right and avoiding the copies is hard.

Anyhow, good luck if you decide to look into it more.

markmaker, 2023-10-14:
Hi @ian-arkver,

> Simply replacing opencv with a cuda enabled version won't achieve any CUDA use or speedup.

Duh! I always (naively) thought the OpenCV API would abstract the hardware from the user/developer, i.e. that it was a matter of the platform, whether certain routines run on the CPU, its various vector-instructions, or the GPU.

Now I see they made it [a separate module](https://opencv.org/platforms/cuda/). And then even vendor specific! 

So lame! What is the purpose of such an abstracted API, if it then doesn't abstract anything? Very disappointed.

_Mark

ian-arkver, 2023-10-14:
Hi Mark. Yup, and a different data type too (not Mat)

markmaker, 2023-10-14:
> If you end up copying between a Java VM allocated buffer and an nvidia nvhost allocated NVMM buffer and/or Cuda Device Memory then all the benefit of CUDA acceleration will probably be lost. 

Not so sure about that. Of course it would be better not to, but then again, those algorithms that really matter, are many orders of magnitude slower than the copying, even taking into account it would now happen across PCIe. 

Plus it would be some work but doable to change the Pipeline into using `mat`s more seamlessly. 

_Mark

ian-arkver, 2023-10-14:
I guess it depends on the res & framerate and the algo in question. My experience is more with multiple 1080p60 or 2160p30 streams where the memcpy burden can mount up rapidly. You'll never (rarely?) get rid of copies entirely, but minimising is always good.

It's been a long time since I looked at OpenCV, so I can't recall whether Mat is a base class of GpuMat. I've also only looked at it in C++, so dunno whether the Java bindings put further restrictions on. You might (or might not) want some kind of abstract buffer handle type to hide different types behind.

Nano/Xavier/Orin are unified memory system SoCs, so no PCIe transfers needed. If you get it right all you need to do to switch between CPU accessing a buffer and GPU accessing it is tell the caches (flush/invalidate). In the more general case though, CUDA will be on a discrete nvidia GPU, so yeah, PCIe.

elfarolab, 2023-10-14:
I really like these techie insights and conversations, feeling like within family..

Opencv 4.8 comes with barcode identification. It would be useful for project and warehouse management. It is still the messy part of my lab even if Inventree is helping a lot.

Do you know if there is an easy way to test my custom opencv jar library in openpnp? Substituting the stock version? How can I do that?

Thank you all.

ian-arkver, 2023-10-14:
Not really sure - not tried it, but something like this might be what you're after, assuming you're building OpenPnP from source...
https://stackoverflow.com/a/22300875

elfarolab, 2023-10-14:
Yesterday I understood I need to mess with maven to make it works. I will try. 

Thank you for the URL link, didn't find it yesterday.

👍