openpnp/openpnp #1599: Feature Request: Auto feeders (like Photon) should be able to preemptively feed before pick

andrewortman, 2024-01-28:
# Feature Request
## Describe the Feature
With LumenPNP + Opulo Feeders on OpenPNP, the head does not start moving until a feed operation completes (which may take a second or two). This wastes a lot of time, especially with larger jobs. This feature would allow Photon Feeders (or any feeder) to start a feed operation and allow the head to move into position while the feed operation completes. 

## How Will It Look?
No user interface changes would be required.

I hacked together a patch, and it worked quite well: https://youtu.be/QA5PsO0E8gA. Notice that the feeder motors start moving while the head is moving. This reduces time significantly from the time the head starts to move to when the pick occurs. Often, the pick can happen immediately if you are working with 0402's, as the pitch of the parts is much smaller so the feed motor does not have to run as long. I have been running this code for about a month now without issue, but it is definitely a hack and want community feedback on implementation details before I start working on a proper PR.

## Why Do We Need It?
1. What problem does it solve?
Specifically with auto feeders, this will save time whenever a feed operation can take place while the head is moving into place, instead of sequentially. 
3. Is it useful for everyone, or does it only solve a problem for your machine?
The LumenPNP is most affected by the issue, but any machine with an auto feeder should be able to take advantage of this feature.

## Provide Examples
Most PNP feeders are much faster than the LumenPNP feeder, being able to feed within half a second. However, the current state of the lumen feeder is that it may take a second for larger parts.

andrewortman, 2024-01-28:
My initial idea is to patch the `Feeder`/`AbstractFeeder`/`ReferenceFeeder`/`PhotonFeeder` with a new `prefeed` callback, that allows the feeder to optionally perform some communication / action before the head starts moving. For `PhotonFeeder`, this would call the feed operation over the bus. In `PhotonFeeder`, in the `feed` callback, a polling operation will check the PhotonFeeder via the `MoveFeedStatus` command to see if the latest feed is complete. If it is, the `feed` callback will complete early and allow the JobProcessor to move onto the picking step. 

To limit scope, I can also only update ReferenceAutoFeeder, and ensure PhotonFeeder extends that class instead of the more generic ReferenceFeeder.

If this sounds reasonable, I'd appreciate a thumbs up from the maintainers before I get started on a PR :) If you have other implementation preferences, please let me know!

markmaker, 2024-01-28:
This is already solved for the normal AutoFeeder using the **Feed before Moving to Pick Location** switch and combining it with the right Actuator Machine Coordination, i.e. make the motion **not** wait for the actuation, so they run in parallel.

https://github.com/openpnp/openpnp/wiki/Motion-Planner#actuator-machine-coordination

Ask the User Group for other users of AutoFeeders how to configure exactly (this is just off the top of my head).

Note: this only works if the controller can actually do that. Usually feeders are on separate controllers, so parallelism is guaranteed by OpenPnP, but if I remember correctly, with the Lumen, it is on the _same_ controller. However your "hack" seems to confirm there is at least some parallelism possible.

_Mark

andrewortman, 2024-01-28:
Thanks Mark! I did not know about that! Let me play around for next few days and see if I can get something working without changing source outside of the PhotonFeeder impl

andrewortman, 2024-03-13:
Just an update, I have been busy with other life things, but I was able to get a version of this working. It does a head move inside of the feeder code (like referenceautofeeder does) - this works, but seems a little weird (meaning, is it really the feeder's responsibility to move the head into position?). Unfortunately, there seems to be an additional pause and rotation move above the feeder, I need to figure out where that is coming from because it slows things down quite a bit. I might be able to pick up on this soon, but wanted to share my learnings in the meantime

markmaker, 2024-03-13:
> but seems a little weird (meaning, is it really the feeder's responsibility to move the head into position?)

To optimize it, yes.

> Unfortunately, there seems to be an additional pause and rotation move above the feeder

Yes, but you can optimize this, at least if you have free-running tubes:

https://github.com/openpnp/openpnp/wiki/Nozzle-Rotation-Mode

Use the **MinimalRotation** mode.

From a broader perspective: OpenPnP does not currently _look ahead_ to optimize its operation. This would be a tall order to implement in the JobProcessor. There have been repeated discussions in the group about this, though, so perhaps this will come some day. Everybody is invited to contribute! :star_struck: 

_Mark

andrewortman, 2024-03-14:
Perfect! I have next few days off, and want to run a few jobs. will try to measure times before and after the change :) once Im happy I'll put an MR up for ya'll to review

markmaker, 2024-10-01:
Assuming no news is good news.

closed 2024-10-01 by markmaker