openpnp/openpnp #1603: Support for Desktop Smt Pro?

brentpicasso, 2024-03-03:
Hi, I learned about the Desktop SMT pro https://desktopequip.com/desktop-smt-pick-and-place-machine/ that claims compatibility with OpenPnP.

I don't see it in the documentation - is this supported natively by OpenPnP? I am wondering if they forked / customized the standard build.  I'm waiting for more in formation from them.  Thanks1

mcix, 2024-03-04:
I have one, we are currently setting it up to use with OpenPNP. It is taking us some time to figure it out, the supplier has some document on how to set up the machine.

OpenPNP is also new to me so that was expected

I have not placed any parts with it, the camera looking up worked initially but now it's no longer recognized so we are getting support on this..

brentpicasso, 2024-03-04:
Hi @mcix  thanks for the reply.  I'm still waiting from them for more help. 

Are they having you use the standard openPnP software from this repo, or have they forked to another repo, or have they made closed-source customizations? 

Could you upload the setup instructions here as a PDF?

brentpicasso, 2024-03-04:
I have been emailing with them and they are providing a custom version of OpenPnP that has not been added to this repo. I've asked them on the status of when it will be added here.

@mcix are you using their customized build or generic OpenPnP?

markmaker, 2024-03-04:
I hope they are not violating Open Source licenses. They must publish all custom versions in source code. Please help us enforce this.

_Mark

brentpicasso, 2024-03-04:
I got a download link to a custom version of OpenPnP, and I got the impression the code they provided was source. 

I offered to help get it integrated into the repo here, and then it could be listed as a supported system.

brentpicasso, 2024-03-04:
Here's the link to all of the supporting docs and their download of openPnP. 

https://drive.google.com/drive/folders/1yMz2hUjKqQYJ8zTXSMPBcsmlUP0nB7Oi?usp=drive_link

They seem friendly and I'm guessing they would want to contribute to OpenPnP with some guidance from the community.

brentpicasso, 2024-03-05:
@atanisoft did the package contain source code? I have linux, so I'm not able to look at the .exe.

markmaker, 2024-03-05:
The guidance is all there:

https://github.com/openpnp/openpnp/wiki/Developers-Guide

They should know.

_Mark

atanisoft, 2024-03-05:
@brentpicasso I'm also on Linux, however I poked around the link and I'm fairly confident they are violating all terms of the open source license for OpenPnP. They are hosting only an executable which appears to be OpenPnP binaries but not source (I didn't download to confirm as I have zero trust with random binaries on a google drive account).

I would suggest sending them a request for their modifications and to have them comply with the open source license of OpenPnP.

brentpicasso, 2024-03-05:
Thanks, i will.

brentpicasso, 2024-03-06:
Hi, I'm still communicating with them on it.  Here's what I wrote:

```

brentpicasso, 2024-03-06:
Please let me know if you'd like to be included on the email thread.

RichardTHF, 2024-03-29:
@brentpicasso did you ever hear back from them? I have just tried speaking to them on Taobao, and they seem uninterested in supporting anything other than Windows 10, but I would rather use MacOS or Linux for my controller, rather fustrating.

RichardTHF, 2024-03-29:
So I reached out to them via Taobao, and their response was they plan on releasing the build for it in the 3rd quater of 2024, which obviously goes against the licence. 

This is very dissapoinitng, and I regret buying this product. It has extremely poorly written documentation, no support, and no way to run it with anything other than Windows 10. I wouldn't advise purchasing this machine until the above is fixed, and personally if I was OpenPNP I would point out that they are in violation of their licence agreement.

RichardTHF, 2024-03-30:
@brentpicasso doubt that will make much difference, they seem to give zero shits. I got a chinese colleague of mine to also contact them on Taobao and they just ignored it.

brentpicasso, 2024-03-30:
I've asked my contact to speak to the 'boss' of the company, we'll see what happens.

brentpicasso, 2024-04-08:
Not yet. I will ping again.

vtjballeng, 2024-04-12:
There's a variant or rebrand here with the PickSoEasy 3000 or PSE-3000. I suppose there's a chance Maurizio has the source and may be a simpler point of Contact.
 https://www.picksoeasy.com/pages/PSE3000.html 

Hi @brentpicasso !

brentpicasso, 2024-04-12:
@vtjballeng Hi! Good to know. Can you message me their direct email so we can chat with with them? Perhaps that will break the logjam here.

markmaker, 2024-04-14:
Thank you for staying on top of this! :+1:

vtjballeng, 2024-05-08:
> @vtjballeng Hi! Good to know. Can you message me their direct email so we can chat with with them? Perhaps that will break the logjam here.

It's on their [contact page](https://www.picksoeasy.com/pages/contact.html) @brentpicasso:
mobile phone: +86 15988459270
Feel free to call. Maurizio
pickme@picksoeasy.com

mikhailmihalkov, 2024-06-10:
Hello everyone... I want to buy this machine. Is there a way to work with it in a linux environment?

mikhailmihalkov, 2024-06-11:
> > @vtjballeng Hi! Good to know. Can you message me their direct email so we can chat with with them? Perhaps that will break the logjam here.
> 
> It's on their [contact page](https://www.picksoeasy.com/pages/contact.html) @brentpicasso: mobile phone: +86 15988459270 Feel free to call. Maurizio [pickme@picksoeasy.com](mailto:pickme@picksoeasy.com)

Good afternoon! Any progress on this issue?

JacobChrist, 2024-06-12:
@mikhailmihalkov This a issue list for OpenPnP software (not a specific machine).  OpenPnP can control many different machines and will run in Linux.

mikhailmihalkov, 2024-06-12:
> @mikhailmihalkov This a issue list for OpenPnP software (not a specific machine). OpenPnP can control many different machines and will run in Linux.
I'm trying to understand what's the difference between the original and the program for Desktop SMT pro

JacobChrist, 2024-06-12:
@mikhailmihalkov The Desktop SMT pro looks like a machine that was built to work be controlled by OpenPnP.  What do you mean by the "original"?  If you have questions about the about the Desktop SMT pro you should contact the company that makes it.  A link appears in the first entry on this issue list.

mikhailmihalkov, 2024-06-12:
> @mikhailmihalkov The Desktop SMT pro looks like a machine that was built to work be controlled by OpenPnP. What do you mean by the "original"? If you have questions about the about the Desktop SMT [...]

How does the manufacturer's file differ in size from https://openpnp.org/downloads/ I contacted the manufacturer and received software for Windows but not for Linux. Now I'm trying to understand if it is possible to run this machine on Linux.

brentpicasso, 2024-06-12:
Desktop SMT pro is offering a customized version of OpenPnP, in the form of a windows exe. 

They are not offering the source code of their customizations, which is in violation of OpenPnP's license.  

If their source code was available, there would be a chance to compile it for Linux, and possibly fold their changes into this repo as a machine variant support. 

My recommendation is to use a carrot / stick approach to get them to release their source code. 

I'm not going to be purchasing this system any time soon, so someone should take this up with Desktop SMT's owners to inspire/compel them to do the right thing.

mikhailmihalkov, 2024-08-11:
> I have one, we are currently setting it up to use with OpenPNP. It is taking us some time to figure it out, the supplier has some document on how to set up the machine.
> 
> OpenPNP is also new to me so that was expected
> 
> I have not placed any parts with it, the camera looking up worked initially but now it's no longer recognized so we are getting support on this..
> 
> Edit: the machine uses a duet2 board (clone)

Did you manage to get it running?

e-joerg, 2024-08-30:
I have a SMT-Pro here in germany. It arrives one week ago from china.  
Mine uses the Duet FLY CDY V3. I not had known that they are using a cusomized version.
I had gone through the calibration process they are advising and it works so far. But it is very poor documented - and it is taking a while. 
I not have test a real board right now, nor placed any part. A friend of mine also here in germany uses the machine with good success.
Mine is running on Win11 without any trouble. 
I guess mauritio (picksoeasy.com) uses the original version of OpenPNP. There is a comprehensive video on setup the machine in usig original OpenPnp on his youtube channel. 
I woud prefer to do a change back to OpenPNP but im not really deep in this software righ now.
Just had seen there is a setup-Wizard - and it colides with the custom setup. You can see some entanglements I was wondering about, but not had figured out more...
regards joerg

markmaker, 2024-09-18:
# :warning: Please do not buy these machines, until they comply with the Open Source license promptly and continuously. :warning: 

_Mark

mikhailmihalkov, 2024-09-19:
> # ⚠️ Please do not buy these machines, until they comply with the Open Source license promptly and continuously. ⚠️
> _Mark

You will also prohibit buying any equipment in China and produce it yourself, then tell us about licenses for open source programs and the letters need to be made even larger so that Mr. Xi can see it.

markmaker, 2024-09-19:
@mikhailmihalkov,

I don't understand what you want to say, sorry.

Certainly, it is not about buying in China or any other specific nation, but about complying with Open Source Licenses. They benefit from the years of OpenPnP development for free, so now they need to give back. Other Chinese companies do comply, and they do prosper. 

The others we will at least shun, by not providing support, by telling people not to buy their machines.

_Mark

WainingForests, 2024-09-19:
@mikhailmihalkov There's two openpnp kits / machines sold from china commonly used on the google group. Well liked. I doubt there was any meaningful hardship to those creators to follow openpnp guidelines and the benefits that openpnp provides likely outweigh the time loss easily. Don't try and conflate being respectful to the code base / work put in as restrictive or targeted.

brentpicasso, 2024-09-19:
@WainingForests Interested in these well liked openpnp kits.  Could you provide the model / links please?

WainingForests, 2024-09-19:
@brentpicasso 

Microsmt's creator is the same individual who produced the poorman feeder that is been used for years, as well as the newer bing feeder (moving away from the 0816 push rod design). https://www.microsmt.com.cn/

Panda Placer is the other. Of note I own a mid 2023 model Panda Placer and it's worked great for me. https://pandaplacer.com/
Leo I've talked to quite a few times, he's been super helpful and even will share 'in the pipeline' changes to demo if you ask.

I wager both machine options would work amazing for their cost. Though neither compete with entry industrial machines which can be had for only 3-5 times the cost. 

There was a bit hostility between the two creators whos machines I've mentioned here on the google group but I don't think that should weight too heavily when comparing the machines.

mikhailmihalkov, 2024-12-03:
> @atanisoft did the package contain source code? I have linux, so I'm not able to look at the .exe.
The code was sent by the manufacturer
https://github.com/lei111/openpnp--

mikhailmihalkov, 2024-12-03:
> > @atanisoft did the package contain source code? I have linux, so I'm not able to look at the .exe.
> > The code was sent by the manufacturer
> > https://github.com/lei111/openpnp--
This is a branch of openpnp, so that openpnp can support the bottom of the dual camera This is not a finished product, because there are too many changes, resulting in some unexpected bugs, but the bottom of the dual camera can be used normally, but because of the need to make a distinction with the normal model, so added a verification mechanism. You can enable the dual camera automatically by authentication openpnp, is a very, very great project, but for newcomers to start is not too friendly. We have moved to building SMT software from scratch, which is more suitable for commercialization. So this branch will not be maintained in the future.

markmaker, 2024-12-03:
> > @atanisoft did the package contain source code? I have linux, so I'm not able to look at the .exe.
> > The code was sent by the manufacturer
> > https://github.com/lei111/openpnp--

There is no code that I see. That's just a bunch of text files.

_Mark

mikhailmihalkov, 2024-12-04:
> > > @atanisoft did the package contain source code? I have linux, so I'm not able to look at the .exe.
> > > The code was sent by the manufacturer
> > > https://github.com/lei111/openpnp--
> 
> There is no code that I see. That's just a bunch of text files.
> 
> _Mark
Apparently these are modification files for assembly ?

mikhailmihalkov, 2024-12-04:
> > > > @atanisoft did the package contain source code? I have linux, so I'm not able to look at the .exe.
> > > > The code was sent by the manufacturer
> > > > https://github.com/lei111/openpnp--
> > 
> > 
> > There is no code that I see. That's just a bunch of text files.
> > _Mark
> > Apparently these are modification files for assembly ?

This is java code, you need a utility to compile it

markmaker, 2024-12-04:
Where do you see Java code?

mikhailmihalkov, 2024-12-04:
> Where do you see Java code?

https://github.com/lei111/openpnp--/blob/main/pom.xml
You need to install the compiler and some other packages, everything is described in the file, this is a standard variation on Java, then everything is assembled and works. I'll check it over the weekend, if everything is okay, I'll post it here for Linux

vonnieda, 2024-12-04:
Hi Mikhail,

That repo is a copy of the openpnp repo with all of the source code removed. The only remaining files are related to build and administration. All of the history has been removed, as well. The source code should be under a directory called src. You can see the difference here between that repo:

https://github.com/lei111/openpnp--/tree/main

And the real OpenPnP repo:

https://github.com/openpnp/openpnp/tree/test

I have not followed this thread closely, but this is a bit concerning. I would recommend to everyone reading that you only use OpenPnP binaries from official sources. Those are:

https://openpnp.org/
https://github.com/openpnp/

Binaries posted to this mailing list are not official, and should not be used.

Thanks,
Jason

mikhailmihalkov, 2024-12-19:
> Hi Mikhail, That repo is a copy of the openpnp repo with all of the source code removed. The only remaining files are related to build and administration. All of the history has been removed, as [...]

The manufacturer sent the source code in an archive
[openpnp-双目.zip](https://github.com/user-attachments/files/18194035/openpnp-.zip)

mikhailmihalkov, 2024-12-20:
Hi guys) If you have any questions about the open source code, ask here. I have compiled a theme version for Linux, I specifically use it for Mint 22. If you have the ability, you can compile a deb package so as not to use the system dependency Java 11.

mikhailmihalkov, 2024-12-20:
Please help if anyone can build a deb package with a built-in Java library. My knowledge was enough to build on Java 11. Thanks for your help)

mikhailmihalkov, 2025-01-02:
Reviews recommendations will be for Linux?

1off-dev, 2025-02-10:
I want to share my terrible experience with the Desktop SMT Ultra and DesktopEquip.com to help others avoid the same issues I faced.

The machine arrived practically without a manual and came with software that only works on Windows. It was sold to me as a ready-to-use machine, but I immediately realized that not even the basic calibrations had been performed. For example, the main carriage was misaligned with the rails and not perpendicular to the working surface, making calibration impossible.

The Chinese contact who sold me the machine assisted me for two hours, trying to fix the first issues I encountered. I had to manually override the limit switch and reposition the carriage.

After that, I attempted to configure the basic settings, but the software proved to be completely inadequate.
	•	Components are randomly misplaced instead of being placed accurately.
	•	Nozzles lower too early and crash into everything.
	•	Component pickup often occurs in the wrong position.
	•	The errors are completely inconsistent—if it were a calibration issue, I would expect repeatable mistakes, but the behavior is entirely unpredictable.

Now, the supplier is completely unresponsive. They do not reply to emails, and my WhatsApp messages are undelivered.

I have collected videos and photos proving that the machine is completely unusable.

I hope this review helps others avoid the same mistake.

mikhailmihalkov, 2025-02-11:
> I want to share my terrible experience with the Desktop SMT Ultra and DesktopEquip.com to help others avoid the same issues I faced.
>
> The machine arrived practically without a manual and came with [...]

The machine you are discussing does not support openpnp.
https://drive.google.com/drive/folders/17S3sNhL7aK6uS5OrPPDx7ibNw51JEEeR

mikhailmihalkov, 2025-09-04:
> I compiled the code in the zip attached here in post ([mikhailmihalkov][on Dec 19, 2024]) and I see the same zip has been added to the Lei111 github linked above (3 weeks ago).
> 
> It works on my Desktop SMT.
> 
> The code seems to be the same at the compiled exe I was supplied by the support for the seller.
> 
> I'm not sure what version it was built off.

Have you managed to avoid system dependency java 11?

covert-, 2025-09-04:
yes, got it running on openlogic 17.0.16 jdk. I have Co-Pilot in Visual Studio and it had a few shots to make it work.

POM changed webcam capture from 0.3.12-SNAPSHOT to 0.3.12 and another change to get nashorn-core 15.4 to get user .js scripts to work.