openpnp/openpnp #1626: Improve "Effective Safe Z coordinate XX.XXXmm is outside Safe Z Zone" message

JacobChrist, 2024-05-27:
# Problem Report
## Describe the Issue
Recently, several people have been confused by the message "Effective Safe Z coordinate XX.XXXmm is outside Safe Z Zone".  This message sometimes appears when you try to pick a part that has a height set to 0. 

## Steps to Reproduce
When trying to "Perform a feed and pick on the selected feeder" from the feeders tab, the machine goes to the right coordinates picks the parts then OpenPnP displays always the same message "Effective Safe Z coordinate 40.000mm is outside Safe Z Zone", refusing to move the nozzle refusing to home the machine displaying the same message.

-----------------------------------------------------------------------

# Feature Request
## Describe the Feature
Simply expanding the existing message to add "Make sure selected part height is not 0.00mm in the parts tab"

## Why Do We Need It?
This message would probably help noobs to OpenPnP save hours of frustration.

markmaker, 2024-05-27:
This should not happen, or at least **Issues & Solutions** should warn you against a misconfiguration. Or maybe it is a bug, so we need to fix it rather than to add a better message. :smile: 

Please post the machine.xml and log of when this is happening.

Maybe it is easier to discuss this on the discussion list.

_Mark

markmaker, 2024-10-01:
Assuming the Issues & Solutions hint helped, having heard nothing since (or me having missed it),  I'm closing this for now.

closed 2024-10-01 by markmaker