openpnp/openpnp #1629: Switching to Field Oriented Control for the X and Y motors is not without problems

FlashPackets, 2024-05-29:
My Lite Placer needs to reliably place 0201 size components. ... lots of them

Not to long ago I had a system that was very near done using the Duet 2. I really like the Duet and ordinarily it would be a good choice. But I couldn't get repeatable performance with the optical fiducial. Not the fault of the Duet 2. Unfortunately, the Duet 2 does not have a socket for the stepper motor drivers making it impossible to use MKS-SERVO42A drivers which provide impressive performance and placement accuracy, even if the accelerations are so high that it would cause a conventional drive to miss steps. A FOC system will instead oscillate around the endpoint until it converges. The point being that using FOC motors eliminates them as contributing to any subsequent placement inaccuracies. Any problem would then be belt tension, bending, binding or loose hardware.

So I switched to a MKS Monster8 controller with 8 socket ports running Marlin with FOC controllers on X & Y. So, life was good, lights are going on and off, the system homes and it demonstrates excellent motor precision. The system response to a M114 was immediate, within the same millisecond according to the log. However, reading in the notes, it is clear that the use of E normally associated with extruders is deprecated and that the firmware should be recompiled with zero extruders and remapping the E to generic linear motors like x,y,z . 

So, Marlin bugfix-2.1.x was recompiled and loaded onto the Monster8.

Using putty as the terminal interface  M114 produces this result:
X:0.00 Y:0.00 Z:0.00 A:0.00 Count X:0 Y:0 Z:0 A:0
ok

G28 homes beautifully with the carriage at the highest Z and downward movement reported as negative.
Perfect!

The version of OpenPnP 2024-02-08_07-03-33.5dd12c7 was compiled from source and appeared to be working just fine a few days earlier along with bugfix-2.1.x as it is "out of the box"..

so ./OpenPnP          brings up the program

2024-05-29 15:58:37.824 Main INFO: Bienvenue, Bienvenido, Willkommen, Hello, Namaskar, Welkom, Bonjour to OpenPnP version 2024-02-08_07-03-33.5dd12c7.
2024-05-29 15:58:37.826 Scripting TRACE: Scripting.on Startup
[DBG ] FOURCC = MJPG
[ERR ] setProperty (ID=1) failed on VIDIOC_S_CTRL (errno 13)
[ERR ] getPropertyLimits (ID=2) failed on VIDIOC_QUERYCTRL (errno 22)
[ERR ] getAutoProperty (ID=2) failed on VIDIOC_G_CTRL (errno 22)
[ERR ] getPropertyLimits (ID=2) failed on VIDIOC_QUERYCTRL (errno 22)
[ERR ] getPropertyLimits (ID=2) failed on VIDIOC_QUERYCTRL (errno 22)
[ERR ] getPropertyLimits (ID=2) failed on VIDIOC_QUERYCTRL (errno 22)
[ERR ] getProperty (ID=2) failed on VIDIOC_G_CTRL (errno 22)
[ERR ] getAutoProperty (ID=2) failed on VIDIOC_G_CTRL (errno 22)
[ERR ] getAutoProperty (ID=5) failed on VIDIOC_G_CTRL (errno 22)
[ERR ] getAutoProperty (ID=5) failed on VIDIOC_G_CTRL (errno 22)
[ERR ] setProperty (ID=4) failed on VIDIOC_S_CTRL (errno 13)
2024-05-29 15:58:38.357 AbstractBroadcastingCamera TRACE: Camera OpenPnpCaptureCamera thread 34 started.
[DBG ] FOURCC = MJPG
[ERR ] setProperty (ID=1) failed on VIDIOC_S_CTRL (errno 13)
[ERR ] getPropertyLimits (ID=2) failed on VIDIOC_QUERYCTRL (errno 22)
[ERR ] getAutoProperty (ID=2) failed on VIDIOC_G_CTRL (errno 22)
[ERR ] getPropertyLimits (ID=2) failed on VIDIOC_QUERYCTRL (errno 22)
[ERR ] getPropertyLimits (ID=2) failed on VIDIOC_QUERYCTRL (errno 22)
[ERR ] getPropertyLimits (ID=2) failed on VIDIOC_QUERYCTRL (errno 22)
[ERR ] getProperty (ID=2) failed on VIDIOC_G_CTRL (errno 22)
[ERR ] getAutoProperty (ID=2) failed on VIDIOC_G_CTRL (errno 22)
[ERR ] getAutoProperty (ID=5) failed on VIDIOC_G_CTRL (errno 22)
[ERR ] getAutoProperty (ID=5) failed on VIDIOC_G_CTRL (errno 22)
[ERR ] setProperty (ID=4) failed on VIDIOC_S_CTRL (errno 13)
2024-05-29 15:58:38.759 AbstractBroadcastingCamera TRACE: Camera UpCam thread 36 started.

I don't think the above is all that important as the window has both top and bottom cameras working.

Now pressing the little green button hangs the system with 

2024-05-29 16:16:36.729 ReferenceMachine DEBUG: setEnabled(true)
2024-05-29 16:16:39.745 GcodeAsyncDriver DEBUG: serial://ttyACM0 commandQueue.offer(G21, 20000)...
2024-05-29 16:16:39.746 GcodeAsyncDriver DEBUG: serial://ttyACM0 commandQueue.offer(G90, 20000)...
2024-05-29 16:16:39.746 GcodeAsyncDriver DEBUG: serial://ttyACM0 commandQueue.offer(;, 20000)...
2024-05-29 16:16:39.746 GcodeAsyncDriver DEBUG: serial://ttyACM0 commandQueue.offer(M114, -1)...
2024-05-29 16:16:39.746 GcodeAsyncDriver$WriterThread TRACE: [serial://ttyACM0] >> G21
2024-05-29 16:16:39.746 GcodeDriver$ReaderThread TRACE: [serial://ttyACM0] << k
2024-05-29 16:16:59.747 GcodeAsyncDriver$WriterThread TRACE: [serial://ttyACM0] >> ;
2024-05-29 16:17:39.747 MessageBoxes DEBUG: Error: java.lang.Exception: GcodeAsyncDriver timeout waiting for response to M114
2024-05-29 16:17:43.925 AbstractMachine TRACE: Machine entering idle state.

Hung for over a minute

For the record the POSITION_REPORT_REGEX is
^.*X:(?<X>-?\d+\.\d+) Y:(?<Y>-?\d+\.\d+) Z:(?<Z>-?\d+\.\d+) A:(?<A>-?\d+\.\d+).*
that tests as being correct with the captured putty sequence above.

Everything appears to be correct, but it doesn't work. It behaves  like a broken regex ...

Why would recompiling Marlin to produce what amounts to changing a single character is a M114 response have such a profound effect?

I've no idea what to do to fix this!!

tonyluken, 2024-06-08:
This probably would have been better addressed in the OpenPnP forum rather than here as an issue.

Anyway, the problem appears at the second of these two lines:

2024-05-29 16:16:39.746 GcodeAsyncDriver$WriterThread TRACE: [serial://ttyACM0] >> G21
2024-05-29 16:16:39.746 GcodeDriver$ReaderThread TRACE: [serial://ttyACM0] << k

OpenPnP received a "k" when it is expecting an "ok" in response to the G21 command. It is hanging waiting for the "ok" which it never receives. Check your serial comms - perhaps your baud rate is not correct?

markmaker, 2024-10-01:
Seems solved.

closed 2024-10-01 by markmaker