openpnp/openpnp #1635: Tape specification

hydra, 2024-06-02:
# Feature Request
## Describe the Feature
Have a system to allow tape specification to be entered, so that the information can be re-used by the feeders.

That is, a part has a 'package', a package comes in packaging (tubes, reels of tape, etc) and the feeders use parts in packaging.
e.g.  
for a 910ohm 0402 resistor from Vishay with a part code of CRCW0402910RFKTD
'Package': D10/CRCW0402 - https://www.vishay.com/docs/20008/dcrcw.pdf
Packaging: Vishay P0/RT7 - https://www.vishay.com/docs/20014/smdpack.pdf

The specification for this is as follows:

[image]

## How Will It Look?
Add a UI to allow tape specification to be entered, with the following required information:
* width
* hole-pitch
* top to hole-center y offset
* hole to cavity-center x + y offset
* part pitch

For example for the Vishay part above, in P0/RT7 packaging, the values are:

* width: 8mm (W)
* hole-pitch: 4mm (P0)
* top to hole-center y offset: 1.75mm (E)
* hole to cavity-center x + y offset: 2mm,3.5mm (P2, F)
* part pitch: 2mm (P1)

When defining a 'part', allow the user to pick the tape specification from one of the defined specifications.

Additionally, when assigning a 'part' to a 'tape' you also can have a single place to define the 'rotation in tape', instead of in each feeder, as it does not matter which feeder is used, this rotation is constant.

Feeder orientation could also be defined, per-feeder, so that the single 'rotation in tape' assigned to the part can be applied regardless of where the feeder is mounted or the tape is used.  e.g. left/right/front/back.

## Why Do We Need It?

The UI currently has a String entry for Tape Specification as follows:

[image]

And each feeder system asks the user to fill in details that come from the manufacturers tape specification, for example:

Strip feeder:
[image]

Drag feeder:
[image]

Push pull feeder:
[image]

Blinds feeder:
[image]

Schultz feeder:
[image]

Bamboo feeder:
[image]

There is much duplication of information, all of which comes from the tape specification.
Each of the feeders needs to know which tape is used, and from that most can derive what they need based on the tape specification.

Some feeders might still need to know where the second hole is, but even that can be given a sensible default, e.g. 4mm to the right if the feeder orientation is left-to-right.

2. Is it useful for everyone, or does it only solve a problem for your machine?

it's useful for everyone, and there are industry standards for this: EIA-481 - https://community.nxp.com/pwmxy87654/attachments/pwmxy87654/kinetis/37830/1/EIA-481-2003.pdf

## Observations

There are many duplicated UI elements that could be deleted, which would result in a more consistent UI and a better user experience; especially true when using multiple different feeders, .e.g sometimes on a reel, sometimes in a blinds feeder.

Having to set the rotation in tape only once also prevents issues when the tape is moved to a different feeder.  If you've ever made a batch of PCBs and had to rework it because you moved a tape from a feeder on one side of the machine to a feeder on the other side you'd which this was already implemented.

markmaker, 2024-06-02:
Hi @hydra 

thanks for the spec. This is the way to go :+1: 

First let us talk about the "philosophy" of OpenPnP. I'll present my view (which is just one opinion that other contributors might or might not share) and you can then tell us your vision.

1. The "philosophy" that I think OpenPnP mostly follows, is aimed at the use case of quick prototyping, rather small series production, mostly in a DIY, or early startup, or third-party service oriented small business environment, with maybe sprinkles of exotic fringe production use cases inside larger businesses. The want a few assembled PCBs fast, without any unnecessary ballast, let me call it "bureaucracy".

2. The "philosophy" that everybody else (not matching the above description) needs, is a completely different one. Though I admit that the practical details are mostly unknown to me, I consider it a matter of fact, that everybody that wants to produce even middle-sized series, would be daft not to either use an external assembly service provider, or then buy [professional ultra-fast and precise machines and their (arguably more important) feeder ecosystem](https://youtu.be/tJ-N-6DMIe4) and matching professional software, with their tight integration into larger development, production and enterprise interfaces. Even though they cost tens of thousands of dollars, they will quickly pay themselves back.

Between the two, it is a matter of preparation time vs. unit time. 

In the first world with OpenPnP we have very(!) low numbers of units. Therefore unit time is less important and we can afford to use simpler, slower machines. Relatively speaking, preparation workload becomes much more important. In development/prototyping you are constantly changing your projects, use new or changed parts, etc. Or when you assemble small runs for many different clients, you want your "data interface" to be very simple and independent from their design software, and you still want to bang their PCBs out without much preparation hassle. 

I believe OpenPnP excels in that world, it runs from a minimal set of placement coordinates, auto-learns new parts and packages, adds new feeders quickly, and off you go. Your data does not provide a part height? No problem, we can measure it on the first use of that part, [using "Auto-focus"](https://github.com/openpnp/openpnp/wiki/Up-looking-Camera-Auto-Focus). Your data does not provide a footprint? No problem, the bottom vision is agnostic towards your parts, and mostly just works. But you can provide footprints for _the very few_ parts where it is required, i.e. [parts larger than the camera view (multi-shot vision)](https://github.com/openpnp/openpnp/wiki/Vision-Compositing). You want your feeders quickly setup? In the simplest form, use the [BlindsFeeder](https://github.com/openpnp/openpnp/wiki/BlindsFeeder), it detects most if not all of its properties (you circled in red) automatically from "green screen" computer vision. Or [use one-click setup of the ReferencePushPullFeeder](https://github.com/openpnp/openpnp/wiki/ReferencePushPullFeeder#the-next-feeder), after a first one of its kind has been defined, it clones all properties from the first one initially and whenever you want. Want to be quick and dirty about Z coordinates of feeders and the PCB? Just use a [Contact Probing Nozzle](https://github.com/openpnp/openpnp/wiki/Contact-Probing-Nozzle). Want to use cheap DIY belt machine, and mount PCBs and shoddy feeders with hand accuracy? No problem, use fiducials for homing, fiducials on PCBs and sprocket holes on feeders on the fly. Want to be tolerant about not so accurate feeder picks, cheap motion hardware etc.? Simply use bottom vision on each part, and adjust for all static and dynamic errors there.

In the second world, it is the other way around. You have large volumes and the unit time is all that matters. You'll want your machines to run large series mostly unattended. But this means that you have to get your production much more reliable, as any issues sum themselves up quickly to be intolerable. Worse, any _mistake_ will be multiplied into all of your PCBs thousandfold, with no chance of reworking them (as you would with a prototype, or tiny series). You'll want your placement data, parts in feeders etc. to be absolutely right, which means integrating data throughout the whole operation. But once you start (unattended) volume production, you can take your time to prepare for the next run, and consequently be happy with all the "bureaucracy" required to ensure the higher quality standard. Conversely, most of the "auto-learn" and "readjust" automatism, I described above, would be prohibitive in these scenarios, as simply too unreliable, too late in the process, not feeding back e.g. into the E-CAD etc., and/or simply way too slow during production.

Of course there is a gray area between the two worlds. OpenPnP can be a stepping stone from the first to the second world, e.g. for a startup. And yes, the payback-point where the stepping-up makes sense, does vary with the salaries of your employees in your country, comparing the cost of more manual work to the price tag of a professional machine. So I'm not against adding more functionality to create an overlap. But not at the expense of the original "philosophy".

Having said all that, I hope you see the implications for this PR:

[... cut ...]
In any case, all such Tape Specification must remain optional. Therefore, I don't see how UI elements can be deleted.

But in case you use them, should they follow reference semantics rather than copy semantics, so one can quickly choose from a small list? Does this mean adding a new main-level tab with list in the UI? Do we want that added complexity? Or should it be copy semantics and therefore much more work to enter? And where would the data be? On each package, or on each vision settings?

Taking all this into account, this becomes quite a difficult PR.  Challenging enough? :grin: 

_Mark

GeorgeIoak, 2024-06-02:
I just thought I'd add a comment since I've been somewhat silently watching @hydra adventure recently as I also am trying to bring up a LumenPnP.

@markmaker It's very encouraging to hear your philosophy as I 100% agree with your comments. I've been involved in manufacturing for a very long time and leave the *real* manufacturing to the experts, my CM. My hand soldering skills are not great so I thought I'd get something to help with prototypes.

I'm not sure this PR would address the issues that I've faced but some of these auto routines don't always work (or maybe user error or some level of it's not always going to work).

- I've used Altium for a very long time and just generated PnP files and threw them over the fence and let the assembly engineers deal with it (I also provide ASCII CAD data). Altium doesn't provide a separate pick location to be defined, you either get the calculated centroid, or the 0,0 reference of your footprint but I doubt anyone who is designing a footprint is thinking first of the best location to pick the part). I mention all this because I was trying to work with a FFC connector and the ReferenceStripFeeder and it only would want to go to the center of the pocket to pick the part and I couldn't define an offset so it would grab on the body of the connector and not partially on the latch. Maybe this is something for tape specification or maybe in the feeder or maybe it's just me. I finally gave up and switched to a tray feeder where I could just set the pick location.
- I had the auto setup for the ReferenceStripFeeder work correctly many times but other times I struggled a bit on setting the feed count and part pitch values in order to get the pick location to be correct. I think in this case a strip definition would definitely help.
- Along these lines are the part rotation in the tape. TBH, I'm still a bit confused as to how that should be set since the CAD data has no idea how the parts are in the tape nor does it care but since I'm still confused perhaps a tape definition would help bridge that gap between the PnP file and how that part is picked and placed?

markmaker, 2024-06-02:
> It's very encouraging to hear your philosophy as I 100% agree with your comments.

:grin: 

I don't suppose this PR will address any of the issues you describe:

- I guess we [discussed](https://groups.google.com/g/openpnp/c/btYwYPx6zlk/m/0vqiHMQUAAAJ) the pick location vs. bottom vision center limitation in the group, right?
 
- The ReferenceStripFeeder has its issue. There are several reasons I made the [BlindsFeeder](https://github.com/openpnp/openpnp/wiki/BlindsFeeder). But I'm biased :wink: 
 
- Part rotation in tape should be explained here:
    https://github.com/openpnp/openpnp/wiki/BlindsFeeder#rotation-in-tape

GeorgeIoak, 2024-06-02:
Yeah, I almost didn't comment as I wasn't sure if my findings were relevant to this PR or not.

I saw that post but now that I've learned more I'll go back and re-read it.

The BlindsFeeder makes so much sense. Now that I've messed with the *standard* ones I've come to not like them for a few reasons.

Maybe I'll bring up the rotation again in Discord but in general the answer I get from people is trial and error which just seems wrong. Like you mentioned, for me, this is a tool to try and make a few boards at a time so reducing/relying on setup are most important to me.

Now back to the PR ...

hydra, 2024-06-03:
Thanks for the feedback thus far.

My use-case for OpenPnP is:
* medium volume just-in-time production (>25 < 250) which isn't economical to outsource due to set-up, delivery and import costs from PCB fabs.
* multiple different projects, many of which share common parts (BOM).
* requirement to be able to build any of the projects at any time with the minimum of machine/feeder/loader reconfiguration.
* de-risking projects, by using just-in-time manufacturing, if a project doesn't sell, or has issues, then the cost of failure (components, manufacturing runs, pcbs) is as low as possible.
* flexibility, able to up-rev a project revision quickly without having to worry about large unsold inventory of lower revision boards.

I'm sure there's a way a UI can be created which meets the 'auto-setup' goals, we just have to think about it.

prusk, 2024-06-03:
Let me add a few thoughts here since I have some work in progress for new feeders.

I also had the thought of connecting the parts to tapes but quickly abandoned it because…
First: while all auto feeders have tapes, not all parts come on tapes. Examples: lose parts, tubes, trays.
We cannot even say for sure that all feeders have tapes because of the vibration ones.
Second: the same part may sometimes come on tape, other times lose. 
Third: the same part can come on different kinds of tape. (I lived to see 0402s on 4mm pitch tape)

For the 3 above, connecting the part to the tape even optionally is logically incorrect in my opinion.
So the only assumption I came to that is yet to be proven false is that all auto-feeders have tapes, therefore I’m coupling the Tape object with a subset of feeders.

With that in mind, it could be useful to have the tape specs defined as reusable objects somewhere. It would even simplify a bit the current vision calibration that relies heavily on the initial pickup location +/- 0.x mm.

For RotationInFeeder - I think this is specific to a part manufactured by a specific company, but even this may be subject to change depending on their internal processes.
Because of this I decided to also define the rotation at feeder level.
Basically all the details about the part and the tape are configured when loading the feeder and then they last for 3000 picks, reason why I like my feeders to stay with the reel, therefore the cheapest auto feeders.

hydra, 2024-06-04:
@prusk yes, I have components here in cut  tape, reels of tape, tubes, bags and boxes, I'm definitely aware of this.  Part to Tape assignment is currently optional and could continue to be optional, even with a proper tape specification UI.

Yes, aware of the different types of feeders, I have drag pin for reels, cut-tape holders and vibration feeders here.

Yes, aware the same component can come in different tape, it can also depend which regional factory made it sometimes, this isn't normal though, but it does happen.

The last point is interesting as it indicates there's potentially a one-to-many relationship with parts and tapes

I suppose you could have the tape assigned to the feeder to at least pre-populate the feeder's UI, and @markmaker also I suppose it would be possible to create a button that allowed you to create a tape from what the feeder vision auto-detected, for those feeders that have auto-detection code.

Yes agree with regard to a list of re-usable objects, probably another tab called 'Tapes' next to 'Parts' would make sense.

Rotation, if you check the EIA-481 there is actually a standard regarding where pin one is in the tape.  And if the part's EDA footprint was available this could actually be done automatically by looking at where pin one is in the footprint and then using the EIA-481 standard to pre-calculate the pin-1 quadrants and the rotation-in-tape value, which I'm sure @markmaker would agree that more 'auto' stuff is better.

The feeders on my machine are non-removable and time consuming to load/change.  Depending on the feeder, feeder orientation, actually unreeling orientation, could be set manually, or detected automatically for bus-based systems where the feeder's bus could have the unreeling orientation configured on it and inherited by feeders on the bus.

markmaker, 2024-06-04:
> I suppose you could have the tape assigned to the feeder to at least pre-populate the feeder's UI, and @markmaker also I suppose it would be possible to create a button that allowed you to create a tape from what the feeder vision auto-detected, for those feeders that have auto-detection code.

Yes after reading both your _correct_ comments, assignment must be to the feeder, not the part. And such a button would of course make sense. 

However, what happens, when the auto-detection happens again?

> Rotation, if you check the EIA-481 there is actually a standard regarding where pin one is in the tape. And if the part's EDA footprint was available this could actually be done automatically by [...]

Opinions about what EIA-481 states or not states are diverse. We had rather heated discussions :sweat_smile:. There is no clear definition, as the tape "standard" only talks about "quadrants" and not about angles, the part "standard" talks about "Part Zero", meaning the rotation angle. And to make matters worse, those "standards" evolved, with certain part types actually changing their orientation in the tape.

But it doesn't actually matter, because **a)** manufacturers keep ignoring the standard (which is somewhat understandable, as it is changing), and **b)** users have their parts drawn wrong in their E-CAD-Library, with some major Open Source libs being among them, and setting bad examples. Even the footprints are no help as there is (at least in general) no standard markup "this is pin 1" (e.g. on diodes, transistors, etc.), and they are often not even drawn/exported in the right order. For some part geometries it is even hard to say which is "upper left" ...

So we avoid all this discussion and use the simple method explained in the Wiki: 

1. Whatever your E-CAD lib says is Part Zero is true.
2. We (OpenPnP) _define_ the tape zero degrees orientation as the one with the (round) sprocket holes at the top, simply because the quadrants are illustrated that way in the standards doc. 
[image]

3. Between these two, you have your "Rotation in Tape".

_Mark

GeorgeIoak, 2024-06-04:
I haven't gone back to revisit the Wiki but determining how the part is in the tape and how you have it defined in your library seems somewhat clear. What isn't mentioned though is that when you generate your PnP file from CAD it correctly provides the rotation of the part as is needed on the PCB. I'll go ahead and sound dumb but what didn't seem clear to me was how to add in the orientation of the part in the tape. I see so many references to people just making trial runs and adjusting the rotation in the PnP file so I don't think I'm 100% alone in at least being a little confused.

This is why I decided to add some comments here because perhaps the tape definition would help clarify this or maybe not.

markmaker, 2024-06-04:
> What isn't mentioned though is that when you generate your PnP file from CAD it correctly provides the rotation of the part as is needed on the PCB.

It should be well-defined by the conventions of [right-handed 2D Cartesian coordinate systems](https://en.wikipedia.org/wiki/Cartesian_coordinate_system), which are ~380 years old, so everyone should have gotten it. 

[image]

Some CADs disagree with double sided PCBs, some view the second side from the bottom, some "through", which I can understand. Other than that, any "alternative representation" is just utterly ... dumb (you said the word first :smile:). Even then, the OpenPnP import should fix it for you.

So once imported, this rotation should not have anything to do with **Rotation in Tape**.

>  I see so many references to people just making trial runs and adjusting the rotation in the PnP file so I don't think I'm 100% alone in at least being a little confused.

For **Rotation in Tape** trial and error is a valid strategy for those with bad memory/aversion to RTFM, or (natural) problems with "left" vs. "right" and "clockwise" vs. "counterclockwise", or with negative numbers.

For import files, this is ... Certainly one must have a workflow that gets this right every time. 

If OpenPnP cannot import a common CAD export format right, this should be reported. Yes, I know diptrace (?) fixes are pending due to the CAD changing its export format from one version to the next and not providing the old format as an alternative (which, sorry, is ...).
 
_Mark

closed 2024-06-22 by hydra