openpnp/openpnp #1636: Duplicate UI values for Camera Units Per Pixel.

hydra, 2024-06-02:
# Problem Report
## Describe the Issue
Both the SimulatedUpCamera and ImageCamera have `General Configuration` and `Device Settings` tabs, and there are fields for `Units per Pixel` `X` and `Y`.

I was experimenting with the ImageCamera after making a 5800 x 5300 pixel image with a 0.1mm per pixel ration and found that after changing the X/Y values to `0.1` on the `Device Settings` tab that head moves via the Camera's view still weren't correct until I found the old values on the `General Configuration` tab and applied the same `0.1` value for X and Y there too.

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. Start OpenPnP with default configs.
2. Press On button.
3. Press Home button.
4. Go to the Head's ImageCameraTop.
5. Select 'Device Settings', observe default values `0.042330` for Units Per Pixel X and Y.
6. Change X and Y to 0.1.
7. Download the attached image, open it in the Source Uri. ([image])
8. Press apply.
9. Use the 'Top' camera to jog to the bottom left of the 8mm tape.
10. Observe that it doesn't end up where you let go of the mouse.
11. Click the `General Configuration` tab of the head's image camera.
12. Observe that the Units Per Pixel have the default values of `0.042330`
13. Change the X/Y values there both to 0.1.
14. Press apply.
15. Observe that the Top camera view now jogs correctly when you let go of the mouse.

## Expected Result
a) only one place to set the units per pixel, and it works when you set it.

## Actual Result
* incorrect jogging.
* confused user.
* this bug report. 😉

markmaker, 2024-06-02:
This is exactly as intended. :grinning: 

We want to simulate and test how [Issues & Solutions](https://github.com/openpnp/openpnp/wiki/Vision-Solutions) automatically finds the resolution (Units per Pixel) of the camera. So you as a developer/tester enter them in the ImageCamera as you did, then go to I&S and let it find them out and compare them.

https://youtu.be/md68n_J7uto

Once you use a real camera, physics gives us these and the duplication is gone.

_Mark

ian-arkver, 2024-06-02:
Maybe the Simulated camera's device settings label should be "Simulated Units per Pixel" or something to reduce confusion.

markmaker, 2024-06-02:
Users should always read tool-tips:

[image]

ian-arkver, 2024-06-02:
Indeed. Mine has different text, but I haven't updated my branch in ages.

hydra, 2024-06-04:
> Maybe the Simulated camera's device settings label should be "Simulated Units per Pixel" or something to reduce confusion.

Agreed.  I asked in discord and others there were also confused.

I also find that the tooltips don't always appear for me on both my main machine (Windows 10 and my PnP machine box, Windows 11)  Other apps I use have little '?' indicators that you can click which makes a tooltip appear and which also serve as a call-to-action that there is some help available.

Another way I've seen of providing in-line help is to have a '?' button in the window title, that you can toggle on, and then when you click on any element it shows help/tooltips.

I would welcome any way of improving the discoverability of the UI and reducing confusion.

markmaker, 2024-06-18:
See #1647 that changes the labels as proposed by @ian-arkver. 

Plus I believe the **underlying confusion** is also sufficiently explained.

I never had any problems with tooltips on various Windows and Linux versions. If this persists, please file another Issue, or better yet, try to solve it :+1:. Frankly, I don't see how this can be an OpenPnP issue, i.e. the Java VM should likely be fixed or replaced with a better one.

A "question mark" tooltip marker PR would be welcome, **if** it works generically, i.e. without having to add it explicitly on/near each control that has a tooltip now or in the future.

_Mark