openpnp/openpnp #1644: Incoherent UI experience when creating blinds feeder arrays.

hydra, 2024-06-17:
## Describe the Issue
After reading the documentation, here:

https://github.com/openpnp/openpnp/wiki/BlindsFeeder

I'm unable to create 2 blinds feeder arrays.

I attempted to add 4 feeders, 2 feeders in each array, and assign a 'Feeder group name' of 'A' to the first to, and a 'Feeder group name' of 'B' to the second two.

The entire experience was confusing, which I documented in this video:

https://www.youtube.com/watch?v=kvKI6fJzfZ0

## Steps to Reproduce
See video.

## Expected Result
It's had to say what to expect as the UI is not well designed (IMHO).

## Actual Result
* deletion of existing feeder groups from the drop down.
* loss of fiducial locations.
* confusion, frustration.

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
As I see it, the problem is the use of an editable dropdown in the 'Array' group for 'Feeder Group Name'.

[image]

With know prior knowledge about the implementation, ask your self the following questions:
a. what will happen when you change the dropdown selection.
b. what will happen when you edit the field.
c. what will happen when you edit any of the fields in the 'locations' or 'vision settings' groups.

If you just look at the UI as it currently is, you'll quickly discover that you cannot answer the above questions.

Now add in the 'apply' button, what will clicking it do in combination with a, b and c above?

Now think about what tasks the UI might let you do, without knowing if this list is correct:

1. rename a group name
2. add a group name
3. delete a group name
4. select a group name.
5. change location and vision settings for this feeder.
6. apply location and vision settings to other feeders that use the same group name.

Now look at the UI again, and see if you can think how to do any of those tasks without doing the wrong task.

Now imagine the UI had a non-editable dropdown, but instead had 3 buttons, as follows:

`<plus icon>`
`<red X icon>`
`<pen icon>`

[image]

Now go back to the list of questions above, and you'll see that you can answer questions 1,2, & 3.

You still can't answer 5 and 6 because you don't know what happens when you select a group name (question 4).

If you update the UI so that changing the dropdown /immediately/ changes the  values in the locations and vision settings, then you can answer 4 and 5.

Question 6 is still a mystery however due to there not being any UI element indicating how to do this.  should there be another button?  what does apply do?

More  questions arise now however.

d. what happens if you change location settings for this feeder group, then make a new selection via the dropdown box, should the values persist? should the user be prompted? etc.

It seems to me, that shoe-horning feeder syncronisation into the existing UI, with it's apply button, is currently a bad user experience.

If it was me, I would probably create a separate dialog for editing the locations and vision settings, as follows:

[image]

and

[image]

That way, you always know what's going to happen, and when it will happen.

markmaker, 2024-06-17:
@hydra,

some of your "frustration" comes from the standard Wizard buffered "Apply" system of OpenPnP, and the fact that setters mark the Wizard as "modified". It has confused many users including me. You're extremely welcome to fix it, apply changes immediately, and replace buffers with a (multi-step) "Undo" system instead. I'm personally not JavaSwing expert enough to try.

The next source of frustration comes from the fact that you don't use and configure a real physical BlindsFeeder. The magic happens through vision. [Watch the video](https://youtu.be/dGde59Iv6eY).

At 1:30 of your video you read the Wiki, but then you don't follow it at all. Such irrational behavior will always lead to frustration. As the sentence you read aloud states, you would move the camera to somewhere inside the fiducials where a row of parts are, and press the Auto Setup. This associates the feeder with the array, and (if you had a real BlindsFeeder) detects all the rest of the geometry from vision, it might also detect the OCR/QR label, ergo the whole feeder is automatically setup. No small feat.

Yes, OpenPnP is missing a concept of "feeder group", as separate object, potentially on multiple levels (trolley, group, array, slot, etc.). You're welcome to add such a system and port the existing array, and slot feeders to that system without breaking anything. Change the flat list of feeders into a hierarchy on the UI. Add a function to transform a feeder to another location, group based (translate and rotate, potentially recursively). The community would surely welcome that. But it requires a whole lot of very careful work.

As it is now, each feeder must carry the information of the array too. The info is then synced between the feeders automatically. Yes, I do know that's [not third normal form](https://en.wikipedia.org/wiki/Third_normal_form).

It means that when you were clicking around like a madman you deleted the last remaining feeder with array information for "A", so you lost "A". Completely normal, expected and sensible behavior. Even if there was a separate object for the array, why keep it around, when the last feeder referencing it has gone. In other realms it's called "Sweeping Garbage Collection".

Also note that Array names are optional. A user wanted them and added them. But grouping also works with all feeders on "Default", and then only the fiducials count. I only use it that way.

What you are clicking around, is a completely unrealistic scenario. Nobody would do that with real feeders. Instead you would carefully define each feeder of an array, going through the whole setup, then perhaps move to the next. Its completely natural and the UI is optimized for realistic use cases. 

Use a real BlindsFeeder. Set it up using realistic steps, this time following what the Wiki says. When you then find grave inconsistencies or bugs, report them.

_Mark

markmaker, 2024-06-17:
Ooops wrong button.

hydra, 2024-06-18:
Hi Mark,

Firstly, thanks for taking the time to watch the video and respond.

> some of your "frustration" comes from the standard Wizard buffered "Apply" system of OpenPnP, and the fact that setters mark the Wizard as "modified". It has confused many users including me. You're [...]

Yes, as time passed I might investigate this.

> The next source of frustration comes from the fact that you don't use and configure a real physical BlindsFeeder. The magic happens through vision. [Watch the video](https://youtu.be/dGde59Iv6eY). [...]

Understood, I wasn't able to use the BlindsFeeder auto-setup as I haven't got one to hand, I'm printing one out as I write this and will play with the UI and feedback more when I have.

> Yes, OpenPnP is missing a concept of "feeder group", as separate object, potentially on multiple levels (trolley, group, array, slot, etc.). You're welcome to add such a system and port the existing [...]

There are other feeders that have the concept of a 'bank' or 'group', the PhotonFeeder and Schultz feeders are two that I'm aware of.

I feel that if the concept of a group had been created that the UI experience would be greatly improved.

> As it is now, each feeder must carry the information of the array too. The info is then synced between the feeders automatically. Yes, I do know that's [not third normal form](https://en.wikipedia.org/wiki/Third_normal_form).

I'll respond to this on #1643 as the reply is more appropriate there.

> It means that when you were clicking around like a madman you deleted the last remaining feeder with array information for "A", so you lost "A". Completely normal, expected and sensible behavior. [...]

"madman", heh! 😄 I thought I was going mad it's true.

Let me show you this, from SchultzFeeder

> Also note that Array names are optional. A user wanted them and added them. But grouping also works with all feeders on "Default", and then only the fiducials count. I only use it that way.
>
> What [...]

hydra, 2024-06-18:
> Use a real BlindsFeeder. Set it up using realistic steps, this time following what the Wiki says. When you then find grave inconsistencies or bugs, report them.

See: 
* https://github.com/openpnp/openpnp/issues/1646
* https://github.com/openpnp/openpnp/issues/1645

Also see this video of my second experience of the blinds-feeders UI, attempting to follow the instructions verbatim the resulting failures.

https://www.youtube.com/watch?v=BWz0Ajroj8s

hydra, 2024-06-18:
Other points:

1. I'm not sure if it's intended, but I had some thoughts regarding the 'auto-setup' and it's failure to detect the pocket center-line when using a 'push cover'.

It feels to me, that auto-setup should be disabled or give a helpful error message until the user has picked the 'cover type' and that 'cover type' should not have a default to prevent incorrect setup, or that 'auto-setup' should detect the type of cover.

2. 'open all covers' and 'close all covers' feel misplaced in the UI, they are nothing to do with the 'cover settings'; they are system-wide actions that you need to do, and as such would be better as toolbar buttons on the feeders panel toolbar.

markmaker, 2024-06-18:
See the other [response](https://github.com/openpnp/openpnp/issues/1646#issuecomment-2176088144). Applies here too.

A constructive discussion about feeder groups can be opened in a new issue.

closed 2024-06-18 by markmaker

hydra, 2024-06-18:
Can you please respond to the issues regarding:
* my thoughts on how to improve the auto-setup feature.
* the placement of the 'open all covers' and 'close all covers' buttons.

> A constructive discussion about feeder groups can be opened in a new issue.

yes, I'll create an issue with my thoughts on how to improve the UI, I had some discussions earlier in discord about it and have some useful feedback and notes.

Do you plan on making /any/ changes to the UI and blinds feeder code or safety based on feedback in this issue and the other linked issues?

markmaker, 2024-06-18:
> my thoughts on how to improve the auto-setup feature.

Read the [response](https://github.com/openpnp/openpnp/issues/1646#issuecomment-2176088144). This discussion is moot.

> the placement of the 'open all covers' and 'close all covers' buttons.

I'm reluctant to impose a BlindFeeder specific feature on all OpenPnP users. Maybe if it automatically appeared only if BlindsFeeders were present.

_Mark