openpnp/openpnp #1645: Blinds feeder 'open cover' crashes nozzle into feeder.

hydra, 2024-06-18:
## Describe the Issue
Pressing 'Open cover' after changing the blinds feeder 'feed count' causes incorrect head movement.
The push nozzle goes down, into the closed cover, then the head moves to the start of the feeder, then it attempts to push the cover.

## Steps to Reproduce
1. Setup a single 'push cover' type blinds feeder.
2. Feed a blinds feeder a few times so the feed count is non-zero.
3. Insert a new tape of components, close the feed cover.
4. Reset the 'feed count'  of the feeder to '0'.
5. Press 'feed'.

## Expected Result
Cover opened.

## Actual Result
Nozzle crash into closed cover.

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue

Here's a video of the issue:

https://www.youtube.com/watch?v=XC9ONoppyDM

markmaker, 2024-06-18:
This is not helpful.

There is a big button called **Reset**, that both resets the feed count and sets the cover state to "unknown":

[image]

When you _manually_ set the feed count to zero instead, or any other number for that matter, and press **Accept** you're doing _just_ that. It has no effect on the cover state, why should it? It makes perfect sense to modify the feed count on an already open feeder when e.g. after parts fell out, or after you populated some (expensive) discarded parts back in. You would probably complain the other way if it _did_ reset the cover state.

The BlindsFeeder always has to trust the user to not tamper with the cover state, unless they press **Reset** or home the machine, or start a job which all prompt discovering/asserting the cover state if configured it that way.

_Mark

P.S. this will be my last reaction until you answer [the open question re the ReferencePushPullFeeder](https://github.com/openpnp/openpnp/pull/1641#issuecomment-2175571289).

markmaker, 2024-06-18:
This is no issue in my book.

closed 2024-06-18 by markmaker

hydra, 2024-06-18:
It's an issue in my book, a nozzle crash is a nozzle crash is a nozzle crash and they should always be prevented.

Rather than closing the issue immediately, perhaps it's better to wait for feedback on your response as I have some ideas on how to improve it based on what you said.

1. it's not clear, especially when the tooltips don't work, what the reset button does, since it is a group called 'tape settings', and makes users think that it will reset /all/ their carefuly configured tape settings.  Or perhaps it should be called 'reset feed count' or perhaps feed count and 'reset' should be in a UI group by themselves.

2. If the feed count is made lower than the current feed count, and the cover type was 'push cover' then it should reset the cover state?
3. Perhaps if the cover-state was displayed in the UI, I don't see it, then it would be more obvious that manually changing the 'feed count' does/does-not change the cover-state.
4. It's not clear if there's additional steps required to update the feeder state when feed count is changed to prevent a nozzle crash, this could be improved.

markmaker, 2024-06-18:
Read the [other answer](https://github.com/openpnp/openpnp/issues/1646#issuecomment-2176088144). This is all fall-out from this.