openpnp/openpnp #1646: Blinds feeder feed nozzle crash for 'push cover' due to unconfigured/incorrect centerline.

hydra, 2024-06-18:
## Describe the Issue
When the 'auto-setup' feature does not detect or set the 'pocket centerline', and you attempt to feed, the push nozzle will crash into the edge of the blinds feeder.

## Steps to Reproduce
1. Fit a 'push cover' type blinds feeder, the one I printed has 2x 8mm tape rows.
2. put some 0402 tape into it, i was using 0402 tape with 2mm pitch.
3. Create a blinds feeder in openpnp
4. set the fiducials.
5. calibrate the fiducials.
6. click 'auto-setup'
7. note the 'pocket centerline' and 'pocket pitch' were not detected.
8. set 'pocket pitch' and 'pocket size' to `2`.
9. press 'feed'

## Expected Result
The nozzle should push the cover from the center of the tape.

## Actual Result
Nozzle crashed into the edge of the blinds feeder.

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
* Since the spocket holes are detected and the offset of them from the fiducial is known, a push line can be determined (i.e. usually 3.5mm from the sprocket holes) for an 8mm tape.
* If the tape width is also known, then the center of the tape can be calculated, however the blinds feeder doesn't have an option to set the tape width.
* The feed operation should be disallowed if a 'push-line' is not configured (either by the user failing to do it, or via auto-detect failure).

There are also three 'lines' of interest:
1. push line
2. pocket center line
3. pick line

The 'push line' is not the same as the 'pocket-center-line', the 'push line' should be the center of the tape.
The pocket center line is the center of the pockets
The 'pick line' would usually be the center of the pockets, but for some connectors the pick-line might need to be offset from it.  For instance I have USB connectors where the pick position is NOT the center of the pockets.

markmaker, 2024-06-18:
Breaking my own promise of not reacting. This is just too much. :sob:

You're deliberately playing at being daft. Please stop it!

Read the whole Wiki, it introduces the required positioning on the camera for the **Auto Setup** button in section  [Auto Setup of the Tape](https://github.com/openpnp/openpnp/wiki/BlindsFeeder#auto-setup-of-the-tape) (surprise, surprise). 

Or just read the tooltip:

[image]

Plus, I also already explained the centerline camera position to you [in my previous response](https://github.com/openpnp/openpnp/issues/1644#issuecomment-2173972615).

>  you would move the camera to somewhere inside the fiducials where a row of parts are, and press the Auto Setup

And please print a **blinds cover**, not a sliding cover. You can't follow the Wiki for the blinds cover with a sliding cover.

The sliding cover is just a subaltern alternative for parts that aren't spaced far apart enough on the tape. Operation is clearly much more manual, as [explained in the video](https://youtu.be/dGde59Iv6eY?feature=shared). And [it isn't even documented yet](https://github.com/openpnp/openpnp/wiki/BlindsFeeder#advanced).

It is called "BlindsFeeder" for a reason. Use it as such.

And, for heaven's sake, use green filament, or any vibrant (high saturation) color (a color that does not occur on your parts). Otherwise, the "green-screen" effect can obviously not work right and vision will be disrupted by the parts in their pockets. 

All this should be obvious. I don't believe for a second that you're _really_ that hard on the uptake.

_Mark

markmaker, 2024-06-18:
Closed as not being helpful.

closed 2024-06-18 by markmaker

hydra, 2024-06-18:
> Breaking my own promise of not reacting. This is just too much. 😭
> You're deliberately playing at being daft. Please stop it!

honestly, I'm not, this is just how I'm experiencing it!

> Read the whole Wiki, it introduces the required positioning on the camera for the **Auto Setup** button in section [Auto Setup of the Tape](https://github.com/openpnp/openpnp/wiki/BlindsFeeder#auto-setup-of-the-tape) (surprise, surprise).

I read the wiki with regards to the center-line, and searched for it, but all I found was 'OpenPNP will automatically recognize the blinds in the cover and determine the tape centerline, the pocket pitch and the pocket size'.

> Or just read the tooltip,

again, tool tips don't work reliable on either of my two windows machines (one is Windows 10, the other is Windows 11).  Sometimes they work, but other times when I hover over elements nothing appears. When something's unreliable you often stop using it.

> Plus, I also already explained the centerline camera position to you [in my previous response](https://github.com/openpnp/openpnp/issues/1644#issuecomment-2173972615).

I don't any explanation in the comment you linked.

> And please print a **blinds cover**, not a sliding cover. You can't follow the Wiki for the blinds cover with a sliding cover.

yes, but it's not obvious, and I'm just reporting my experience.  The SCAD file called 'BlindsFeeder-3DPrinting.scad' has tape definition for 8mm tapes with push covers, this is what I used.

> The sliding cover is just a subaltern alternative for parts that aren't spaced far apart enough on the tape. Operation is clearly much more manual, as [explained in the [...]

yes, but the 'BlindsFeeder' supports 'Push Covers' and push covers are in the scad files as mentioned above.

> It is called "BlindsFeeder" for a reason. Use it as such.

How about renaming it to something more suitable, or pulling out the push functionality into a different feeder and call that 'PushBlindFeeder'?  Or just accept that it's called BlindsFeeders and there are different types of 'blinds', a push cover is still a form of 'blind' and know that users will use things you create and report issues they find.

> And, for heaven's sake, use green filament, or any vibrant (high saturation) color (a color that does not occur on your parts). Otherwise, the "green-screen" effect can obviously not work right and vision will be disrupted by the parts in their pockets.

I've already ordered green filament which I'm aware will be far more useful for the 'blinds' type cover and for the vision than for the 'push' type cover.

> All this should be obvious. I don't believe for a second that you're _really_ that hard on the uptake.

it's not obvious, if it was I wouldn't have spend 2 days investigating it, and reporting issues as I find them.  If I run into these issues, it's likely others will too.

Of note, I am a visual learner, if that goes any way to explaining things.  https://www.google.com/search?client=firefox-b-d&q=visual+learner

There is room for improvement of this feeder and I don't feel this issue should just be closed as 'not helpful' when it clearly is to anyone else encountering the same problem as I have until the issue I raised has been resolved in code.

markmaker, 2024-06-18:
> I don't see any explanation in the comment you linked.

It is clearly there and I even quoted it for you. If this a "visual learner" impairment, then why not watch the video?

https://youtu.be/dGde59Iv6eY

It shows you step by step how it works. No reading required. Clearly it is shown how the centerline is derived both for the blinds cover [5:03](https://youtu.be/dGde59Iv6eY) and the push cover [8:06](https://youtu.be/dGde59Iv6eY) where I specifically mention it. Yes, I know the UI is slightly different with two tabs.

A bit of patience is required, though. Please watch the whole thing to really understand.

> yes, but it's not obvious, and I'm just reporting my experience. The SCAD file called 'BlindsFeeder-3DPrinting.scad' has tape definition for 8mm tapes with push covers, this is what I used.

I just checked. There are four blinds covers and two push covers for variety in the example. The first three are all blinds covers.

[image]

And I checked the Wiki. It already explains the difference in the [intro](https://github.com/openpnp/openpnp/wiki/BlindsFeeder#features-and-use-cases):

> - Two cover types: blinds cover and push-away cover.
> - Blinds cover for tapes where part pockets are spaced out (e.g. most passives on paper carrier tape). The cover, having openings at the same pitch, can be shifted by half pitch to close up the parts. Uses almost no extra space on the machine table.
> - The push-away cover can be used for tapes with tightly spaced parts (e.g. larger parts in embossed plastic carrier tapes). The cover will be pushed away from the parts as they are used up. Uses extra space for the pushed-away cover.

And the video also makes it very clear.

Whatever the difficulties: it is one thing to ask for help. It is another to publish public videos on YouTube throwing around allegations about how completely unusable the solution supposedly is, and how much better everything could be done,  but in reality not appearing to _(or rather not wanting to?)_ understand even the basics.

_Mark

markmaker, 2024-06-18:
> until the issue I raised has been resolved in code.

What issue? 

As far as I see, there is nothing left once you remove your mixing-up cover types, and not centering the camera on the centerline before pressing **Auto Setup** (for the first and any subsequent feeders) from the equation. 

Unless otherwise noted there, the same is true for almost all the other Issues you opened.

Maybe I'm mistaken. Do it right "by the book" or "by the video", and if a problem still remains, I'll gladly reopen the Issue(s).

Regarding the misunderstandings that led to all this: suggestions for how to clarify the Wiki are always welcome. Not least because I'm formulating all this in a foreign language, and surely a native speaker could do it more spot-on.

However, I have to say, the "problem solving methodology" you demonstrate in your videos, I cannot take seriously. You are aimlessly clicking around, and trying things, as if this was a fuzzing test or something, and the goal was to deliberately go against the manual or video, to see what happens, and if you could break something. We do not need to cover that kind of erratic behavior in OpenPnP. In fact, the feeder took the onslaught with flying colors. :rofl: 

Many users have successfully used the BlindsFeeders before you. Sometimes they have a question, someone answers, problem solved. Documented in the (searchable) discussion group, or augmented in the Wiki for posterity. Rarely there's a true bug behind it, we fix it, solved. 

First stop? Always the discussion group!

_Mark