openpnp/openpnp #1648: Blinds feeder - 'Open all covers' - 'No feeders found to open' error.

hydra, 2024-06-18:
# Problem Report
## Describe the Issue
Firstly, I apologize for yet another issue on the blinds feeders today, I'm just reporting my experience and this is hopefully constructive feedback, but it is feedback nonetheless.

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. Create a blinds feeder.
2. Set the cover type to 'push cover'
3. Complete all remaining required configuration of the feeder.
4. Verify the 'feed' operation opens the 'push cover' blind once each time 'feed' is commanded.
5. Press 'Open all feeders'

## Expected Result
The cover to be opened to expose the first component at the first pick position.

## Actual Result
An error message "No feeders found to open".

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue

Video of the issue:
https://www.youtube.com/watch?v=l3drv45UkWg

Potential solutions include, but not limited to:
1. disabling the 'open cover' and 'close cover' buttons for 'push cover' type feeders.
2. having the 'open all covers' perform the same operation that 'open cover' does for each push cover.

markmaker, 2024-06-18:
LOL. This is a one-way conversation, it seems. 

Again, read the [response here](https://github.com/openpnp/openpnp/issues/1646#issuecomment-2176088144).

Open/Close semantics (and much else) only apply to **blinds covers**. It's called **Blinds**Feeder for a reason! 

The push cover in contrast is only opened it a bit further on each feed. Unlike the blinds cover it cannot be closed again. Opening the cover for all feeders would expose parts regardless of _when (and if)_ they are used.

_Mark

markmaker, 2024-06-18:
> disabling the 'open cover' and 'close cover' buttons for 'push cover' type feeders.

Users can manually fully close the push cover. This is the assumed position after machine homing. Then you can use **open cover** to open it to the current feed count position (exposing the already picked parts). because this makes no sense in real operation (the next feed will still have to open it a bit more), it is not included in the **open all** function.

You're right that the **close cover** button currently does nothing. It should at least set the internal cover position to "unknown", so it can be pressed after the user manually closed it. 

The "unknown" state is always assumed as "fully closed", and the nozzle would then open up it from all the way back on the next feed.

_Mark

closed 2024-06-22 by hydra