openpnp/openpnp #1661: Improve branch names and download pages

vonnieda, 2024-08-07:
Some users accidentally download OpenPnP v1.0 (out of date, unsupported) when they should download v2.0 (Current, Stable). I think it's time to mostly retire the OpenPnP v1.0 downloads, and update the branch names to reflect how we're actually using them.

I'd like to propose renaming some branches, and fixing some download terminology. I'm opening this for discussion and once we agree I'll open a PR as I think some small code changes will be needed as well as a bunch of admin changes.

Currently:
- Branch **develop** is "OpenPnP 2.0", the recommended stable download.
- Branch **master** is "OpenPnP 1.0", the legacy downloads.
- Branch **test** is "Test Downloads", the test downloads.

Proposed:
- Branch **develop** changes to **main**, and the "OpenPnP 2.0" downloads become the "Stable Downloads".
- Branch **master** gets tagged **v1.x** and deleted, the "OpenPnP 1.0" downloads get moved to a separate page like the "Test Downloads" 
- Branch **test** now merges to **main**, instead of **develop**, but otherwise no changes. Test Downloads remain linked from the "Stable Downloads".
- Change the "Welcome to OpenPnP 2.0" dialog to reflect a general welcome rather than a welcome from 1.0 to 2.0.
- Update the installer to use the changed branch names for updates.
- Update the Actions scripts to use the changed branch names.
- Ensure that updates will have continuity from the old branches to the new. Can use website aliases if needed.
- Probably several other tweaky little things here and there that use the branch names.

Thoughts?

Thanks,
Jason

ian-arkver, 2024-08-07:
Seems like an excellent plan to me.

jbussmann, 2025-01-23:
Only one thing I would change: 

> Branch test now merges to main, instead of develop ~,but otherwise no changes~ and _becomes_ develop instead.

But 🤷‍♂️

vonnieda, 2025-02-02:
This work is now (mostly) complete. The only thing left is to change the updates-develop.xml to point to updates-main.xml and I'm going to wait on that for a few days. This will give us a little time to make sure the new releases are working before automatically moving everyone over. 

I'll comment on the list about this as well.