openpnp/openpnp #1677: Dragpin Order of Operation Change

Ferdinand-33, 2024-09-27:
# Feature Request
## Describe the Feature
Allow Order of operation change with DragFeeders and PushPullFeeders.
Im running OpenPnP on a Charmhigh CHM-T36VA. The tape advance is done by a solenoid actuator mounted on the head that pushes a metal pin downwards into the tape holes, the head then drags the tape forward a set amount (4mm for example), the back off operation is then performed (0.25mm for example), then only is the solenoid turned off and the pin is allowed to return to its neutral state under spring load.
What occasionally happens is the DragPin does not always insert perfectly into the centre of the tape hole due to the tape moving around in the feeder or the tape not always providing the same resitance when pulled, when this occurs the dragpin is often still in contact with the edge of the tape hole after the back off operation and the spring does not provide enough tension for the dragpin to return to neutral before the machine starts the pick operation resulting in the pin ripping the tape, crashing into the machine and breaking the pin.
So my suggestion would be to add a tick box that allows you to select between turning the solenoid off before backoff is performed or after backoff is performed. This will allow the pin to retract at any point during the backoff operation.
Current order of operation: Solenoid on -> Head drags tape -> Back off -> Solenoid off -> DragPin returns to neutral under spring load.
Alternate order of operation: Solenoid on -> Head drags tape -> Solenoid off -> DragPin can return to neutral during backoff under spring load -> Back off

## How Will It Look?
If the feature will include changes to the user interface, how would they look? Include drawings or mockups if possible.
Add a chech box to the DragFeeder and PushPullFeeders that allows the dragpin to be turned off before the backoff command is performed.

## Why Do We Need It?
1. What problem does it solve?
     It will greatly reduce the amount of times that the DragPin gets stuck in the tape feeding hole causing the machine to crash and 
     broken DragPin along with ripped tape.
3. Is it useful for everyone, or does it only solve a problem for your machine?
    All the drag feeder based machinces will have the same issue because of how their dragpins function.

## Provide Examples
Is this something that exists in other pick and place software? Provide examples and references if possible.

Ferdinand-33, 2024-09-27:
What would also help and be usefull is to add the option for a time delay after the solenoid has been turned off but before the backoff is performed

markmaker, 2024-10-01:
> What would also help and be usefull is to add the option for a time delay after the solenoid has been turned off but before the backoff is performed

That can and should be added to the G-code of the drag pin actuator.

Look up `G4` or analog for your controller. 

_Mark

markmaker, 2024-10-01:
I'm just speaking for the ReferencePushPullFeeder, as I believe it is already the goto-solution for any drag feeder.

> So my suggestion would be to add a tick box that allows you to select between turning the solenoid off before backoff is performed or after backoff is performed. This will allow the pin to retract at any point during the backoff operation.

Yes, a way to configure the ON and OFF of the two actuators (or possibly more) would be nice. It would probably need a new tab, with all the actuator-related stuff, in order to not overburden the other tabs' UI.

> when this occurs the dragpin is often still in contact with the edge of the tape hole after the back off operation and the spring does not provide enough tension for the dragpin to return to neutral 

If I understand your failure scenario correctly, you would have to back-off _further_ to make a difference. Otherwise the pin would still be jammed the same way, right?

> the back off operation is then performed (0.25mm for example)

Frankly, as long as your back-off is less than the sprocket hole diameter = 1.5mm, i.e. where the pin could start touching the opposite edge, I don't see how this new feature can solve your problem. You can just increase the back-off to say 0.5mm or 0.75mm (half sprocket hole), and then release the solenoid. I don't see how it could then still be jammed against the edge. 

Conversely, when only backing off 0.25mm, I don't see how releasing the solenoid early could help with the scenario where the pin is still jammed against the sprocket hole edge, i.e. it would only have been jammed _harder_ against the edge in any position further forward, the spring force even more inadequate.

You could drive the back-off at full speed, which will perhaps help to shake the pin lose, and not use up much more time than 0.25mm. Also ask other drag feeder users in the discussion group how they do it, they _are_ using the ReferencePushPullFeeder successfully.

So when I say this Issue is valid, it is more aimed at _other_ scenarios, other feeder types, electrically controlled feeders, cover foil takeup motors etc. and adding more actuators. As such it has come up before, and is valid :smirk: 

PRs are welcome.

_Mark

LorenControls, 2025-02-26:
In this regard to avoid nozzle destruction it is needed to stop the head move when the drag pin is stuck in the tape.This can be done by a firmware  change that  enable the move only when the dragpin position sensor
signals its return to the retracted position. By my experience the machine control has several problems regarding operational security.