openpnp/openpnp #1684: Reference PushPullFeeder test error

chen70854022, 2024-10-21:
# Problem Report
## Describe the Issue
I want to add a ReferePushPullFeeder. I followed the video tutorial to add and set it up, but an error occurred during the material retrieval test, as shown in the picture. After some testing, I still cannot locate where the problem is,By the way, I did not enable OCR

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. Start OpenPnP
2. Create a ReferePushPullFeeder and set it up
3 Click to push one from the selected Feida and retrieve the material

## Expected Result
Pull the ribbon and retrieve the material

## Actual Result
An error occurred when the suction nozzle reached the set Z height downwards

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
[image]

markmaker, 2024-10-21:
Seems you have created a `ScriptActuator` instead of a plain `ReferenceActuator`. And I guess you haven't set the **Script Name** yet?

https://github.com/openpnp/openpnp/wiki/Script-Actuators

_Mark

chen70854022, 2024-10-21:
> 看来您创建的是 a 而不是 plain 。我猜你还没有设置 **Script Name**？`ScriptActuator``ReferenceActuator`
> 
> https://github.com/openpnp/openpnp/wiki/Script-Actuators
> 
> _马克

Thank you very much, it is working now

closed 2024-10-21 by markmaker