openpnp/openpnp #1721: Hybrid stepper motor

karl2996, 2024-12-24:
Hi,
I was wondering if anyone could help me. I need to replace the stepper motor on the z-axis. It is a small nema 23 . I want to replace it with some motors with more torque. Is it possible to use a [hybrid stepper motor](https://www.oyostepper.de/category-1-b0-Hybrid-Schrittmotor.html) only on the z-axis? I am using a CS-Labs CSM10/IP-M motion controller and Mach3. Thanks

yapnp, 2024-12-24:
You can connect any motor driver (open/closed loop stepper, servo, etc.) to your motion controller via the Step/Dir lines. 
But your controller is designed for Mach3, which cannot be linked to OpenPnP without developing special communication drivers.

tonyluken, 2025-01-14:
This doesn't appear to be an issue related to the OpenPnP project. Please use the [forum](https://groups.google.com/g/openpnp) if you need advice.

closed 2025-01-14 by tonyluken