openpnp/openpnp #1747: Zoom behaviour in camera view

jbussmann, 2025-02-09:
The zoom functionality in the camera view is implemented in a linear/additive manner roughly like so:
`zoom = zoom + mouseTicks * zoomIncrement` where `zoom` is valid in a range from 1 to 100.
This is very ill suited because an increment of 1 changes the zoom level anywhere from 1% to 100%. It would be much more useful to implement zoom in a logarithmic/multiplicative fashion like so:
`zoom = pow(zoom,  mouseTicks * zoomIncrement)` where `zoomIncrement` obviously needs to be a number larger than 1.
This would result in a much smoother and even zooming experience, especially on cameras with a large FOV, where a substantial part of the available range from 1 to 100 is effectively used.

tonyluken, 2025-02-09:
I don't see how your proposed equation would work. When zoom = 1, how would it ever change since 1 raised to any non-zero power is still 1? Perhaps you meant `zoom = zoom * pow(zoomIncrement, mouseTicks)`?

jbussmann, 2025-02-10:
You are of course right, this is how I meant it to be. I was a bit in a hurry and therefore glanced over two other things that might or might not be of importance.

- The code implies that if `renderingQuality == RenderingQuality.BestScale` is `true`, then `zoom` can only be incremented in integer steps. I don't understand why this is but since it initially increases `zoom` by a factor of 2, the proposed method would yield an initially equivalent behaviour if `zoomIncrement = 2`.

- Currently the zoom increments range from 0.001 to 10.0. Is there a reason for such extreme values? They both seem useless to me. Additionally, I don't see the current zoom value displayed anywhere so I argue that those numbers are meaningless to the average user and words like large/course, medium and small/fine would be better suited. As for the specific values of `zoomIncrement` I propose the middle three of the currently five values, simply incremented by 1.0.

jbussmann, 2025-03-02:
Adressed in #1766.

closed 2025-04-18 by jbussmann