openpnp/openpnp #1809: UI becomes unresponsive after running for long periods

jwilo, 2025-05-10:
When running jobs continuously for several hours, at the approx. 2-3 hour mark the OpenPnP UI becomes extremely unresponsive, and requires a close/restart of OpenPnP.

Task monitor shows nearly 100% CPU, and high memory usage.

This is reproducible every day.

[image]
[image]

tonyluken, 2025-05-14:
I suggest you try the latest Test version of OpenPnP. It has a lot of memory leaks fixed in it that may help you with your problem.

jwilo, 2025-05-14:
Thanks Tony, I'll install that and see how I get on as I'm currently running for 6 hours a day or so right now, so should be able to see the significant improvement.

jwilo, 2025-06-05:
So far, better for sure. I unfortunately had the finish a long run prior to installing the test version and have had limited use since - but in the next few weeks I should see more workload on the test version.