openpnp/openpnp #1811: Nozzle position problem during jobprocess

blspro123, 2025-05-11:
Hi,

My new machine is up and running, but sometimes, randomly, fatal accidents occur . The Z-axis does not lift the nozzles to a safe position before moving the X and Y axes during the pick or place action. The attached picture on the right side shows the operation when the Z position changes; on the left side, the moveTo command is missing, along with the related G-code. To avoid any crashes, I set a safe Z coordinate for the parts and the board and ran the job with many variations without a real picking and placing. I encountered problems when N1 changed to N2 and back, primarily. Investigating the log, the result was always the same as presented in the attached picture.

[machine.pdf](https://github.com/user-attachments/files/20150137/machine.pdf)

[log1.txt](https://github.com/user-attachments/files/20149999/log1.txt)

[image]

The machine based on microsmt construction
Controller: BIGTREETECH SKR-PRO-V1.2
OpenPnp version is 2025_02_02_20-41-38.e468422, 
OS : Windows 10 Enterprise 2016nLTSB 64-bit
Processor: Intel Atom x5-Z8350 1.44GHz
RAM: 4GB

tonyluken, 2025-05-14:
Ok, you should first ask for help in the [OpenPnP Forum](https://groups.google.com/g/openpnp) before creating an issue.

Please set your logging level to TRACE and repeat your test. Then post the entire machine.xml and the entire log file from that session.

blspro123, 2025-05-14:
Hi,

Since the reported test, I installed version OpenPnP-windows-x64-test
2.3_2025-05-10_16-10-14.8add581, re-adjusted some parameters of N1 and N2:
Safe Z from -5mm to -12mm, Dynamic Safe Z on. Re-adjusted HOME_COMMAND, by
Betz Technik First I tested with a safe height setting (nozzle should not
collide with anything), then with the actual setting, so that an actual pnp
operation could be performed. It worked without errors with 20-30 parts
placed. What still happens is that after BottomVisionSetting, sometimes the
nozzle does not raise to the safe height.
I attached the requested files.

Regards

tonyluken ***@***.***> ezt írta (időpont: 2025. máj. 14.,
Sze, 19:41):

tonyluken, 2025-05-16:
>I attached the requested files.

Where???

blspro123, 2025-05-17:
I attached it to the message I received in the email. I'll post it here too.

[OpenPnP.log](https://github.com/user-attachments/files/20263719/OpenPnP.log)

[machine.xml.txt](https://github.com/user-attachments/files/20263726/machine.xml.txt)