openpnp/openpnp #1829: Manual nozzle tip changes do not trigger NozzleTip.Loaded event script

jwilo, 2025-05-18:
**Environment:**

OpenPnP version 2025-01-31_23-28-59.3430bfa
Windows 10

**Description:**

Per this thread https://groups.google.com/g/openpnp/c/iNcQDCCWQzU

**Expected behaviour:**

The NozzleTip.Loaded script to be called when performing a manual nozzle tip load, same an an automatic load

**Observed behaviour:**

Script is not called

janm012012, 2025-05-23:
Hi jwilo!

This script (as well as NozzleTip.BeforeLoad) are only called in case automatic nozzle tip change is configured. 

Jan

jwilo, 2025-05-23:
Hi Jan,

Thanks for your response.

I wonder if this is the "right" behaviour, for lack of a better word.

.Loaded and .BeforeLoaded make no distinction by name between a manual and auto load, leading to the confusion that raised this ticket.

Perhaps a more predictable behaviour would be to call these on manual and auto changes, or split the event calls into .AutoLoaded, .ManualLoaded etc.

What do you think?

janm012012, 2025-05-23:
Hi JW!

OpenPnP is open source and hence anyone can request and provide modification. I'm the last one to judge about "right" or "wrong". 

What you reported sound to me like a feature enhancement. If you can provide a working version that does not distort existing setups, I don't see a reason not to merge it. Unfortunately I don't have experience with scripting and can't test the changes you're suggesting. I've checked the code and I agree with you, that I also don't see a reason why this scripts could not be called on manual nozzle tip change as well. Initially I thought it shall be easy to just move the two scripts one level up outside the `if (changerEnabled) {`, however on manual load the exception, that informs the user about the loading it thrown at a different level and the function calls itself after the exception has been processed. That makes it confusing where to call the script exactly. If someone can suggest and debug changes, that would make the situation safe.

Btw: renaming scripts is not a good idea because it might break existing setups. You're free to add new scripts or call them at other locations again like adding one before throwing the exception if that's what you're missing.

Jan