openpnp/openpnp #1867: if using `GcodeAsyncDriver` without `DELAY_COMMAND` dwell is masked by move actions

Jorropo, 2025-07-02:
New in 2.3 if `DELAY_COMMAND` is not configured and you are using GcodeAsyncDriver it will fallback to java `Thread.sleep` dwell.

However let's assume there already are some commands sent to the controller, let's continue to say it have 1000ms of travel moves in queue.

Then openpnp tries to dwell, so it call `Thread.sleep(200);` however it does not wait for coordination, so the 200ms sleep runs in parallel of the `1000ms` of travel moves; result you don't dwell (200 < 1000).

---

This can be worked around by configuring `DELAY_COMMAND`, with marlin configure `DELAY_COMMAND` to `G4 P{TimeMS:%d}`.

janm012012, 2025-07-02:
Hi @Jorropo!

Would you please describe the scenario in which you see the issue? I'm guessing from mentioning "dwell" that this is vacuum valve related. Did you checked its coordination? If dwelling is not synchronized correctly, the actuator itself is very likely not coordinated correctly.

Jan

janm012012, 2025-07-02:
Hi Jorropo!

We shall check your logs to to find out whats going on and sync it with your configuration. Please report the issue to the mailing list and post a log at trace level and your machine.xml. This will allow other to join in.

For the pump you need to configure the required power on wait time to establish stable vacuum. The code for that still uses Thread.sleep() so it shall not be affected by executing the delay in the driver.

Jan

janm012012, 2025-07-02:
I'm with you! However, who is generating the delay? The one that's requesting the delay is responsible to also enforce the required coordination. For a pick, you'd command the machine to move to the pick location, then open the valve to establish vacuum and final continue moving. If needed, the vacuum valve actuator can request machine coordination (it usually has to) which would insert the required commands to wait for all effected controllers to finish moving. If a static dwell time is desired, it can be configured and will be executed immediately after the vacuum valve actuator has been actuated. For part-on detection, before-read coordination can be enabled.
What you describe sounds like missing machine coordination. Machine coordination is very basic and practically done everywhere and has been there since the beginning. I actually don't understand why it's not working for you and I need to before applying any modifications to it. I also don't understand why this issue has not been reported earlier as you're very likely not the only one that has no DELAY_COMMAND specified.
You may also follow the discussions in the related PR where we agreed on keep the responsibility for machine coordination at the respective actuators.
Please don't get me wrong. I'm not saying there is nothing, I just need to understand why. By providing a log file (at trace level) I can analyze it.

janm012012, 2025-07-03:
Hi Jorropo!

> Ok so where my assumption was wrong. Thanks for correcting me.

To my understanding vacuum valves need the `Before Actuation?` coordination because the valve might otherwise get opened to early sucking up the part in an uncontrolled fashion or get closed to early dropping the part anywhere. If `DELAY_COMMAND` is defined and the same controller is used to drive the value and all axis, `CommandStillstand` is sufficient. For all other scenarios `WaitForStillstand` is required. While you're on it, you might consider enhancing Issues & Solutions to check this...

> I'll post the machine.xml and logs tomorrow.

I'm looking forward to studding them.

Jan

closed 2025-07-08 by tobydickenson