openpnp/openpnp #1873: Report: The Button of "Edit Placement Feeder" didnot work well.

QUTAOJUN, 2025-07-25:
- The button is in the JobPlacementsPanel. 
- After clicking, the software jumps to a blank interface, which seems to be that the Feeders Panel has not been loaded and is displayed. We need to manually click the Feeders Tab to enter.
- Please forgive me, I am not good at code writing and bug fixing. Just report on problems that have been found :)

[image]

https://github.com/user-attachments/assets/b080fba9-0c87-4f56-8da8-f169541562eb

tobydickenson, 2025-07-25:
Thank you for the bug report.

In a quick test I find that this button is working for me. It opens the expected feeder. Please can you repeat your test with the log set to TRACE level and share the full log file.

QUTAOJUN, 2025-07-30:
> Thank you for the bug report.
> 
> In a quick test I find that this button is working for me. It opens the expected feeder. Please can you repeat your test with the log set to TRACE level and share the full log file.

Hi,
Addendum: The Log in dual TRACE mode do not record any information about this bug. Or rather, nothing was left behind.