openpnp/openpnp #1891: Design Review request: Feeder Management

tobydickenson, 2025-08-29:
Hi all,

This is a request for a design review. I am considering implementing some features that relate to how feeders are selected for a part, and placements retried during a job.

Firstly, this will address some problem behaviors relating to job errors. If the job error handling policy is set to "alert" then the user sees an error message box right away. These problem behaviors occur if the error handling policy is set to "defer". Problem behaviors are:
1. Suppose there are 100 placements of a part. Size checking is enabled, but the configured size is slightly wrong. All 100 placements will be attempted, each one resulting in a vision size check error, and 100 parts are systematically dumped into the discard pile.
2. Suppose there are 100 placements of a part. 99 of them are placed correctly, but one suffered a sideways pick error. This is detected at vision and the part discarded. The placement left as a "pending error" to be shown at the end of the job, without any retries.Those problem scenarios are related. The common factor is that the job processor currently has exactly one attempt at each placement. It would be better to:
1. Disable a feeder after several consecutive errors.
2. Have some retries of each placement within the job processor, provided the feeder is not disabled.

I also want to support the following new use cases:
1. Suppose we have few parts left on the tail of a previous tape; this is too short for an auto feeder so it will be loaded on a strip feeder. We want to use those up first, then automatically switch to using the new reel on an auto feeder.
2. We have two auto feeders set up with the same part. If one feeder is working well but the other has a problem, it should automatically use only the good feeder to continue making progress in the job.
3. We have two auto feeders set up with the same part. If both feeders are working well, the job planner optimiser should automatically consider both options and select the nearest

Implementation plan:
1. Some new configuration state is needed; by default to disable a feeder after 3 consecutive failures, or 2 failures out of 5. Store this in the machine object? or a new persistent object inside the machine?
2. Nozzle currently has an attribute which tracks which Part is loaded on the nozzle. It gains a new attribute to track which feeder the part came from.
3. Feeders receive a notification (TBC from Nozzle? or JobProcessor?) when a part has been successfully placed, or when a part is discarded due to a problem. It tracks those statistics and disables itself when thresholds are exceeded.
4. Feeder gains a new "Priority" attribute; low / normal / high.
5. There are several implementations of a findFeeder method; these need to be rationalised.
6. findFeeder() operates by finding all the enabled feeders, filtering by priority, then allowing the job planner optimiser to select if there is more than one option.
7. Some feeder classes operate singly. Others like SlotAutoFeeder operate multiple feeders collectively. I anticipate these new features working with all feeder types equally.

janm012012, 2025-09-03:
Hi Toby!

This sounds like an ambitious and very helpful improvement! 

> Those problem scenarios are related. The common factor is that the job processor currently has exactly one attempt at each placement.

I would like to clarify that first: the JobProcessor is designed as a state machine consisting of a series of linked steps, that are executed until no placements are to be processed. The current design uses separate steps for picking, aligning and placing parts (and other things). Some of them - like aligning - supports to retry a configurable amount of times but there is no feature yet that supports picking a new part if bottom vision has failed because there is no retry counter to leave the loop.

Some of your intended features could be solely implemented in the JobProcessor like disable-feeder-after-n-failues and would then not inherit any knowledge of previous jobs.

Other features like feeder-priorities, would be a per-feeder attribute, which would need to resist in the feeder. Combined with your working-with-all-feeder-types goal, I'd recommend to refactor the code into a general feeder base class and a feeder specific function class. The new base class could then (later) be enhanced to support a remaining property or provide an interface to an inventory management system or be enhanced to support feed-while-moving. This class could also collect statistics across jobs.

I don't see why the part/nozzle relation shall be extended. At present a nozzle picks a part and the JobProcessor initializes the alignment/bottom vision (if requested). The result is - AFAIK - only available to the JobProcessor and the JobProcessor is - AFAIK - the one that request the discard. Hence the JobProcessor should keep track of the feeders failure rate, likely by storing it in the feeder. 

> The overall purpose of those features is to improve the probability of a job finishing with 100% placements complete without user interruptions.

This seems challenging: with error handling set to defer you'll not see any errors until the job has finished. With your new improvements, you'll likely only cover some errors by providing more then one feeder. If e.g. bottom vision setup is bad, a second feeder will not help. If seems that you're suffering from some errors on a regular base and therefore set error handling to defer but you still wont the job to finish without errors. May I ask what errors that are?

Jan

janm012012, 2025-09-04:
Hi Toby!

> > I would like to clarify that first: the JobProcessor is designed as a state machine consisting of a series of linked steps, that are executed until no placements are to be processed. The current design uses separate steps for picking, aligning and placing parts (and other things). Some of them - like aligning - supports to retry a configurable amount of times but there is no feature yet that supports picking a new part if bottom vision has failed because there is no retry counter to leave the loop.
> 
> Yes. `jobPlacements` is the to-do list. `Plan` makes its selection, and **moves** a couple those placements from `jobPlacements` to `plannedPlacements`. `Pick`, `Align`, and `Place` operate on `plannedPlacements`.
> 
> I am anticipating that this new feature operates by `Plan` leaving `plannedPlacements` unchanged, and the placements only get removed from `plannedPlacements` when `Place` has been successful. In other words, a placement remains on the to-do list until it has been done.

I don't see that: JobPlacements has `status` and plannedPlacements are a selected by this status. In `Plan` placements with status `Pending` are selected and in `Place` the status is changed to `Complete`. If case a `JobProcessorException` is thrown, the status is changed to `Errored` (via setError() in `ReferencePnPJobProcessor` line 2376). If I understand the code correctly, placements status could be reverted to `Pending` to reprocess it (would require an exit path to leave this loop at some point). 

> Openpnp has several layers of retries (including two different mechanisms that are both labelled "pick retry count"!). As part of this development I will add an overview of all these layers of retries to the wiki.

Do you mean the ones in `Feeders` and `Part`?

> > Some of your intended features could be solely implemented in the JobProcessor like disable-feeder-after-n-failues and would then not inherit any knowledge of previous jobs.
> 
> Yes. But JobProcessor is already huge, and the more it can delegate to other classes the better.

Agreed, As the Planner is a separate interface with two implementation/instances it could be easily refactored. Also each step is a separate implementation/instance of 'Step' that could be all refactored into individual files.

> > Other features like feeder-priorities, would be a per-feeder attribute, which would need to resist in the feeder. Combined with your working-with-all-feeder-types goal, I'd recommend to refactor the code into a general feeder base class and a feeder specific function class. The new base class could then (later) be enhanced to support a remaining property or provide an interface to an inventory management system or be enhanced to support feed-while-moving. This class could also collect statistics across jobs.
> 
> I am torn between either: a. What you wrote, or b. Adding all that state to a new class which manages the whole fleet of feeders, gathers statistics, and makes judgements across the whole fleet. In this scheme Feeders remain unchanged; feeders just feed.

Good questions: I just checked the code again and found, that we already have the `Feeder` interface, which is used to implement `AbstractFeeder`, which is extended by `ReferenceFeeder` which is then used to implement eg. `ReferenceStripFeeder`. If you put the error handling configurations into one of them either all Reference Feeder or all Feeder will inherit them automatically. A dedicated wizard would probably be needed to provide a unified UI.

(The old  idea of a remaining counter could be added to this UI as well likely together with some other common parameters.)
(The idea of moving while feeding could be implemented by changing the `Feeder` interfaces `feed` into something like `startToFeed` and `waitForFeedToFinish`. `startToFeed` would default to nothing and `waitForFeedToFinish` to `feed`. If the JobProcessor then calls the two new all feeders would still work as is.)

> > I don't see why the part/nozzle relation shall be extended. At present a nozzle picks a part and the JobProcessor initializes the alignment/bottom vision (if requested). The result is - AFAIK - only available to the JobProcessor and the JobProcessor is - AFAIK - the one that request the discard. Hence the JobProcessor should keep track of the feeders failure rate, likely by storing it in the feeder.
> 
> `Nozzle` already tracks which `Part` it is carrying, and I think there is a close parallel with it also tracking which `Feeder` it got it from.
> 
> Consider this scenario:
> 
>     1. Operator pauses a job
> 
>     2. Operator switches to a feeder tab, and presses the "test pick" button. This is the operator's choice of feeder, not necessarily the same as what the system would have picked
> 
>     3. The part is still on the nozzle when the operator resumes the job
> 
>     4. The job processor `Pick` step [sees the right part is already on the nozzle](https://github.com/openpnp/openpnp/blob/6cef9940b177978b632378bbad4ceba330939472/src/main/java/org/openpnp/machine/reference/ReferencePnpJobProcessor.java#L1291-L1303), so it doesnt actually need to do the pick.

That's a good point! So keep with your idea to extend the nozzle to carry information about the feeder the part has been taken from. (I'm not sure how take-back is implemented: does it try to put the part back into the selected feeder allowing to move parts between feeders feeding the same part? I guess this would not interfere with your ideas, [... cut ...] le the job is running, manually re-enable the feeder that was automatically disabled by this new mechanism, and finish the job with 100% placements complete.

That might not work as the required setup/preparation steps are only executed at job start (at least at present). (The feeders `prepareForJob` is only called as part of the `PreFlight` step.)

Sounds like you actually need the two features disable-feeder-after-x-out-of-y-errors (where y might be optional) and pick-align-retry-counter.

Jan

janm012012, 2025-09-05:
Hi Toby!

> > (I'm not sure how take-back is implemented: does it try to put the part back into the selected feeder allowing to move parts between feeders feeding the same part? I guess this would not interfere with your ideas, would it?)
> 
> Currently it uses the same `findFeeder` logic as fee+pick, so it theory you can transfer from one feeder to another with
> 
>     * enable feeder A / disable B (both configured with the same part)
> 
>     * pick
> 
>     * disable A / enable B
> 
>     * take back
> 
> 
> There will be a new `findFeeders` method that returns a list of feeder options. I suggest it makes sense for take-back to use that, prefer the original feeder if it is still on the list, and use the nearest (or other criteria) otherwise.
> 

I only found `findFeeder` implemented in `AbstractFeeder`. Extending it to `findFeeders` to handle per feeder priorities or using more then one feeder in a job - as you suggested - is IMHO a good enhancements.

TakeBack, however, is IMHO a seldom used feature which shall not hinder you to go forward. If it still works by putting the part back into the feeder it came from or into any feeder (as per today) or to the feeder the user selected, I'd be find.

> > A. This would be covered by detecting a series of bottom vision errors (due to bottom vision retries) and then picking an other part requiring "just" a pick/align/place retry-counter feature.
> 
> I agree a per-placement retry limit makes sense too.
> 

I've to clarify myself: it's actually a pick/align retry counter that would be needed in order to retry on bottom vision failure. (I'm not aware that the place step can fail.) This would be raster simple to implement: just add a counter (per nozzle) maybe to 'PlannedPlacement` that is initialized to zero and incremented for retry until a configurable (global) limit has been reached.

> > Btw: could you detect the side-ways picking using vacuum part-on detection?
> 
> Maybe with a more precise pressure sensor, but not with the sensor I currently have on the machine. Sideways picks block the hole.
> 

That's my experience too. With properly setup feeders, vacuum part on detection is not needed and if not used, it can not fail causing addition disruptions.

> > This would best be covered by adding a general "remainig" or "Max Feed Count" feature. Like with other feeders that already support that, the feeder is just empty once the configured amount of parts have been taken.
> 
> I will add that to the base class (and new wizard) if it makes sense when I look at the code in detail.
> 

Yes, that's probably a different story. It might require moving `Feed Count` as well which would require changing all feeder implementations... Probably a good idea for future enhancements (like inventory management, which would hook into `feed`) but IMHO unrelated to what you're intending to do here.

> > That might not work as the required setup/preparation steps are only executed at job start (at least at present). (The feeders `prepareForJob` is only called as part of the `PreFlight` step.)
> 
> Thats a good catch. I wonder what happens today if you enable a feeder mid-job? I will review the code.

I've never tried. Adding support for mid-job-feeder changing is likely something that could be added separately. Feeders would need to keep track if they have been _preparedForJob_ and prepared before first use.

Jan

tobydickenson, 2025-09-05:
> > Thats a good catch. I wonder what happens today if you enable a feeder mid-job? I will review the code.

It looks like all the current feeders will be fine if they never get their prepareForJob called.

prepareForJob lets the machine do all its OCR and vision calibration (etc) work up-front, and in an optimal movement pattern. But all the feeders have a defensive implementation where they will do the same work in a `feed()` if needed. The common implementation pattern is that this is done in a `assertCalibrated()` method. The API documentation for `feed()` says "do everything needed..." so arguably this was a requirement all along.

jwilo, 2025-09-29:
> This would best be covered by adding a general "remainig" or "Max Feed Count" feature. Like with other feeders that already support that, the feeder is just empty once the configured amount of parts have been taken.

I'm a big fan of adding support for this; we too regularly run jobs with re-reeled parts, where we buy parts 20-100 at a time, each part costing $30 or so, so having support to just halt a feeder when the re-reeled quantity is consumed would be fantastic.

I've read the above discussion several times, but being totally unfamiliar with the codebase means I still have some questions in my mind. One of the main usability issue I come across is on a multi-nozzle machine, if a vision op fails and the part is discarded, the job cannot proceed without discarding all of the other parts on the other nozzles, as attempting to resume the job throws "No part loaded on nozzle".

Is my understanding correct that these changes will resolve that, and after discarding the part (edge pick, end pick, no pick etc) it'll simply proceed to repick that part without needing you to stop the job, thus discarding all the other good parts?